Ultimaker/Cura #1874: print head size, print one at a time

mitchy2233, 2017-05-24:
no matter how many times i try and change the print head size it will not save my changes!! it did it one time but will not do it again!! ive restarted my computer many times!!  also i sure don't like how you can only fit one object on the bed with the print at one time on!! i use cura 15.06 or what ever and was able to fit 4 of the exact same item on the bed with the print at one time!! why did you change this?? or what is going on???  even when i had the correct "head size" i was still unable to print the exact same items!!  ps ive never had a crash before (extruder crash) why did you change such a good thing? lol

Ghostkeeper, 2017-05-24:
I tried this, using the following steps:
* Add a Custom FDM Printer.
* Change Y Min to 15 (up from 10).
* Restart Cura.
* Open up the Machine Settings again.

Now the Y Min still says 15. Seems to work fine? What steps did you take to modify the print head size and when did it go wrong?

Deepfriedchril, 2017-05-24:
I've also had problems with 2.6.0 losing my machine settings after closing Cura and not allowing me to change them after reopening Cura. Dual extruders get removed and forces single.
I've resorted to creating a new profile every time I want to slice something. This gets fairly annoying with dual extruders that also need configuring.

fieldOfView, 2017-05-24:
When this happens, please zip up your entire configuration folder and post it somewhere so we can have a look. Note: the location has changed for Cura 2.6

fieldOfView, 2017-05-25:
Could you check if this also happens if you add a new printer, and change the settings of that printer? I think the issue is with upgraded printers

Deepfriedchril, 2017-05-25:
I took a look in the config folder and deleted everything in there.
Haven't experienced any issues since my last post. I've been able to change the printer and extruder profiles freely.

BagelOrb, 2017-07-21:
So you mean the bug issolved..? :)

closed 2017-07-21 by BagelOrb