Ultimaker/Cura #1875: Floating supports

ChrisRiddell, 2017-05-24:
Windows 10
Ultimaker 2+
Cura 2.6 Beta

with supports enabled there is some sections where the supports are been placed that will make them floating 3+ layers above the bottom even with the support bottom layer setting as 0.1mm

(happens on both normal 0.15mm and fine 0.1mm profiles)

File from https://www.thingiverse.com/thing:2187088

[image]
[image]

Ghostkeeper, 2017-05-24:
> support bottom layer setting as 0.1mm

You mean support bottom distance?

What I see in your screenshot looks relatively normal. There is one layer of distance between support-top and the model visible on the left-hand side of the first screenshot. Support bottom distance seems to be 0.

ChrisRiddell, 2017-05-24:
Sorry yes the support bottom distance

Here is a video showing what I mean 
https://www.youtube.com/watch?v=XXNcijpaGrI

Ghostkeeper, 2017-05-24:
Hmm, perhaps the pieces of support there are too small to print. Could you try setting the support line distance to something very small, just for testing? Like 0.01mm.

You could also try reducing the support X/Y distance, or set Z to override X/Y.

ChrisRiddell, 2017-05-25:
Changing the support line distance done nothing, But changing  support X/Y distance from 0.7 to 0.4 some what worked.

Ghostkeeper, 2017-05-29:
Sorry, I meant Support Line Width, not Support Line Distance :(

ChrisRiddell, 2017-05-31:
Changing Support Line Width to 0.1 still see section of supports been printed in mid air.

Ghostkeeper, 2017-05-31:
Hmm, well that's strange. So it's probably not that the support is too small to print in that bit then.

Is there a way in which we can reproduce this? It doesn't seem to be happening with the construed examples I'm trying.

ChrisRiddell, 2017-05-31:
Hmmm not to sure, I'll uninstall cura, delete the folder in the %appdata% then reinstall just in case my install is doing something weird.

ChrisRiddell, 2017-07-06:
The issue after reinstalling cura seems pretty much fixed only notice one floating support.
[image]

ianpaschal, 2018-08-07:
@ChrisRiddell Can you share the model so I can check if this is still reproducing and which setting is causing it?

ianpaschal, 2018-08-07:
This no longer reproduces in Cura 3.4.1.

closed 2018-08-07 by ianpaschal