Ultimaker/Cura #1877: Ultimaker 3 aborts print without job name without error

jango-fx, 2017-05-24:
When I try to print something without a defined job name in Cura (which seems to be the case by default, if I print from within Fusion360) I don't get an error message within Cura. Neither at the printer. Instead, the printer just displays
> Print finished, Cooling Printcores
immediately after the start, without printing at all.

Only when I explicitly give the job a name, the issue resolves.

Similar issues are described in the forum:
https://ultimaker.com/en/community/39946-print-finished-immediately-after-starting-a-print
https://ultimaker.com/en/community/36103-print-issue-for-using-simplify3d-for-ultimaker3

For once there is no error / message, giving a hint to the reason of the issue, nor does Cura imply anywhere, that you have to have a job name. 
Indeed you can't even see at first glance, that it is missing, since as long as there is no name, there's only a pencil icon floating at the bottom-right-most corner of the interface, with no indication, that there's a text input next to it...

fieldOfView, 2017-05-24:
This should be fixed already in Cura 2.6 beta. Could you check?

fieldOfView, 2017-05-25:
Duplicate of #1703

closed 2017-05-29 by nallath