Ultimaker/Cura #1879: Cura 2.6 Beta - Exception Setting Heated Bed for Ultimaker Original

jaybaugh, 2017-05-24:
To replicate:
Select add printer and select "Ultimaker Original"
Check "Heated Build Plate"
Get exception:

Version: 2.6.0-BETA
Platform: Linux-4.4.0-78-generic-x86_64-with-debian-stretch-sid
Qt: 5.7.0
PyQt: 5.7.1

Exception:
Traceback (most recent call last):
  File "/tmp/.mount_dDM0Km/usr/bin/plugins/plugins/UltimakerMachineActions/UMOUpgradeSelection.py", line 41, in setHeatedBed
    definition_changes_container = self._createDefinitionChangesContainer(global_container_stack)
  File "/tmp/.mount_dDM0Km/usr/bin/plugins/plugins/UltimakerMachineActions/UMOUpgradeSelection.py", line 56, in _createDefinitionChangesContainer
    global_container_stack.insertContainer(-1, definition_changes_container)
  File "/build/2.6/build/inst/lib/python3/dist-packages/cura/Settings/CuraContainerStack.py", line 312, in insertContainer
cura.Settings.Exceptions.InvalidOperationError: Cannot insert a container into Global stack

-------------------------
Thread 0x00007f39879e4700 (most recent call first):
  File "/tmp/.mount_dDM0Km/usr/bin/plugins/plugins/USBPrinting/USBPrinterOutputDeviceManager.py", line 87 in _updateThread
  File "/build/env/inst/lib/python3.5/threading.py", line 862 in run
  File "/build/env/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/build/env/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007f39a4dde700 (most recent call first):
  File "/build/env/inst/lib/python3.5/threading.py", line 297 in wait
  File "/build/env/inst/lib/python3.5/site-packages/zeroconf.py", line 1569 in wait
  File "/build/env/inst/lib/python3.5/site-packages/zeroconf.py", line 1153 in run
  File "/build/env/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/build/env/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007f39a55df700 (most recent call first):
  File "/build/env/inst/lib/python3.5/threading.py", line 297 in wait
  File "/build/env/inst/lib/python3.5/site-packages/zeroconf.py", line 1569 in wait
  File "/build/env/inst/lib/python3.5/site-packages/zeroconf.py", line 1017 in run
  File "/build/env/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/build/env/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007f39a5de0700 (most recent call first):
  File "/build/env/inst/lib/python3.5/site-packages/zeroconf.py", line 946 in run
  File "/build/env/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/build/env/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007f39a6ffd700 (most recent call first):
  File "/tmp/.mount_dDM0Km/usr/bin/plugins/plugins/RemovableDriveOutputDevice/RemovableDrivePlugin.py", line 58 in _updateThread
  File "/build/env/inst/lib/python3.5/threading.py", line 862 in run
  File "/build/env/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/build/env/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007f39a77fe700 (most recent call first):
  File "/build/env/inst/lib/python3.5/threading.py", line 297 in wait
  File "/build/env/inst/lib/python3.5/site-packages/zeroconf.py", line 1569 in wait
  File "/build/env/inst/lib/python3.5/site-packages/zeroconf.py", line 1153 in run
  File "/build/env/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/build/env/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007f39a7fff700 (most recent call first):
  File "/build/env/inst/lib/python3.5/threading.py", line 297 in wait
  File "/build/env/inst/lib/python3.5/site-packages/zeroconf.py", line 1569 in wait
  File "/build/env/inst/lib/python3.5/site-packages/zeroconf.py", line 1017 in run
  File "/build/env/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/build/env/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007f39c899f700 (most recent call first):
  File "/build/env/inst/lib/python3.5/site-packages/zeroconf.py", line 946 in run
  File "/build/env/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/build/env/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007f39c91a0700 (most recent call first):
  File "/build/env/inst/lib/python3.5/threading.py", line 297 in wait
  File "/build/env/inst/lib/python3.5/site-packages/zeroconf.py", line 1569 in wait
  File "/build/env/inst/lib/python3.5/site-packages/zeroconf.py", line 1017 in run
  File "/build/env/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/build/env/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007f39c99a1700 (most recent call first):
  File "/build/env/inst/lib/python3.5/site-packages/zeroconf.py", line 946 in run
  File "/build/env/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/build/env/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007f39ca5bd700 (most recent call first):
  File "/build/env/inst/lib/python3.5/threading.py", line 293 in wait
  File "/build/env/inst/lib/python3.5/threading.py", line 424 in acquire
  File "/build/2.6/build/inst/lib/python3/dist-packages/UM/JobQueue.py", line 86 in _nextJob
  File "/build/2.6/build/inst/lib/python3/dist-packages/UM/JobQueue.py", line 116 in run
  File "/build/env/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/build/env/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007f39cadbe700 (most recent call first):
  File "/build/env/inst/lib/python3.5/threading.py", line 293 in wait
  File "/build/env/inst/lib/python3.5/threading.py", line 424 in acquire
  File "/build/2.6/build/inst/lib/python3/dist-packages/UM/JobQueue.py", line 86 in _nextJob
  File "/build/2.6/build/inst/lib/python3/dist-packages/UM/JobQueue.py", line 116 in run
  File "/build/env/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/build/env/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007f39cb5bf700 (most recent call first):
  File "/build/env/inst/lib/python3.5/threading.py", line 293 in wait
  File "/build/env/inst/lib/python3.5/threading.py", line 424 in acquire
  File "/build/2.6/build/inst/lib/python3/dist-packages/UM/JobQueue.py", line 86 in _nextJob
  File "/build/2.6/build/inst/lib/python3/dist-packages/UM/JobQueue.py", line 116 in run
  File "/build/env/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/build/env/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007f39cbdc0700 (most recent call first):
  File "/build/env/inst/lib/python3.5/threading.py", line 293 in wait
  File "/build/env/inst/lib/python3.5/threading.py", line 424 in acquire
  File "/build/2.6/build/inst/lib/python3/dist-packages/UM/JobQueue.py", line 86 in _nextJob
  File "/build/2.6/build/inst/lib/python3/dist-packages/UM/JobQueue.py", line 116 in run
  File "/build/env/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/build/env/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Current thread 0x00007f39e8f50700 (most recent call first):
  File "/build/2.6/build/inst/lib/python3/dist-packages/cura/CrashHandler.py", line 100 in show
  File "/build/2.6/build/inst/bin/cura", line 44 in exceptHook
  File "/build/2.6/build/inst/lib/python3/dist-packages/cura/CuraApplication.py", line 659 in run
  File "/build/2.6/build/inst/bin/cura", line 70 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 958 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 969 in _find_and_load
  File "/build/env/inst/lib/python3.5/site-packages/cx_Freeze/initscripts/Console.py", line 21 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 958 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 969 in _find_and_load
  File "/build/env/inst/lib/python3.5/site-packages/cx_Freeze/initscripts/__startup__.py", line 12 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 958 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 969 in _find_and_load

ianpaschal, 2018-02-21:
No more work will be done on versions 2.x. If this is still occurring in versions 3.2 or newer, we can re-open this issue.

closed 2018-02-21 by ianpaschal

Ghostkeeper, 2018-03-13:
This was fixed for the 2.6 stable version by this pull request: https://github.com/Ultimaker/Cura/pull/1891
Github didn't close this ticket automatically because we had to cherry-pick the fix to a release branch.