Ultimaker/Cura #1880: Cura 2.6 Beta - Changed project name is not saved

jaybaugh, 2017-05-24:
Using Cura v2.6 Beta

To replicate:
Load a model
Save to file. Note the filename.
Click cancel.
Change the project name by clicking the little pencil next to the project name. (Bottom right of main window.)
Click "Save to file" button.
Note that filename has not been updated.

Ghostkeeper, 2017-05-28:
I'm not able to reproduce this in Cura 2.5 on Windows 8.1. Perhaps this is a regression, or something that happens only on Linux? (I saw on one of your other bug reports that you were using Linux.)

jaybaugh, 2017-06-09:
After clearing my settings and configuration and using the latest 2.6 Beta, I was not able to replicate this either. Closing this issue.

closed 2017-06-09 by jaybaugh

Ghostkeeper, 2017-06-10:
Hmm, curious what could've cause this then. But oh well, let's hope the bug stays down :)