Ultimaker/Cura #1881: Cura 2.6 Beta - Crash after changing material name

leavion, 2017-05-25:
I created a new material then duplicated it. When I change the display name and click Close, Cura crashes (closes). This is on Ubuntu 16.04.2 LTS (Xenial).

Meanwhile I was able to edit the relevant file in ~/.local/share/cura/2.6/materials and it worked fine.

Ghostkeeper, 2017-05-29:
I'm not able to reproduce this. I did:
* Create a new material, "Custom Material".
* Duplicate Custom Material.
* Select the new Custom Material and rename it to Test.
* Close the window.
* Switch to material Test.

No crashes. It seems to work fine.

leavion, 2017-05-29:
Interesting. For me it's reproducible. Are there any files I should attach to this ticket? It may be an issue with the configuration file(s).

Ghostkeeper, 2017-05-29:
You could try attaching a zip-archive containing that folder `~/.local/share/cura/2.6`.

Ghostkeeper, 2017-05-29:
Or just a log file could help a lot as well with where the error is, if it happens to be very obvious.

leavion, 2017-05-30:
Of course now when I try to replicate it again it works just fine. After about 10 times that I managed to crash it before I opened the bug report it now suddenly works. I am not sure why. I will close this ticket now and create a new one if I can replicate it again.

closed 2017-05-30 by leavion