Ultimaker/Cura #1883: Cura 2.6 Beta - z hop no longer functions

notnyt, 2017-05-25:
z hop is no longer working since 2.6 beta

BagelOrb, 2017-05-26:
Do you have Z hop Only Over Printed Parts enabled?

notnyt, 2017-05-26:
Yes, and when it moves over printed parts, it still does not Z hop.  This was working normally in 2.5

notnyt, 2017-05-26:
Nevermind, this looks like it's in the g-code just not displayed in the layers view

closed 2017-05-26 by notnyt

BagelOrb, 2017-05-26:
Non retracted moves are dark blue. Moves which are retracted or z hopped
are displayed in light blue.

<https://github.com/Ultimaker/Cura/issues/1883#issuecomment-304221690>, or mute
the thread

notnyt, 2017-05-26:
was working in s3d a bit where the retract/hop moves show the vertical lift, didn't realize when I was in Cura that they weren't shown