Ultimaker/Cura #1884: Extruder specific start/end Code won't work

ManuelW77, 2017-05-25:
Hey there, Custom FDM Dual Extrusion have a problem, with the Extruder specific setting (start/end Code). They won't used in the gcode. Only the Machine settings start/end gcode will be used.

second thing, what are the variables for extruder 1 and 2 first layer temp and first layer bed temp?

ManuelW77, 2017-05-28:
Can someone confirm this?

I try'd on two different Mac's with blank Cura install.Try'd with adding a second machine, but won't work also. It includes only the Machine specific gcode but not the extruder specific in the output file
.

fieldOfView, 2017-05-28:
I *think* the extruder start code snippet is only inserted when switching from one extruder to another, ie typically not at the beginning of a gcode file

Ghostkeeper, 2017-05-28:
Indeed, it's inserted here: https://github.com/Ultimaker/CuraEngine/blob/b2d998dc65d6085e54ecb701a23a6a49f28db204/src/gcodeExport.cpp#L859

That function is only called here: https://github.com/Ultimaker/CuraEngine/blob/b2d998dc65d6085e54ecb701a23a6a49f28db204/src/gcodeExport.cpp#L882

The setting name for the nozzle temperature on the first layer is `material_print_temperature_layer_0`. The setting for the bed temperature on the first layer is `material_bed_temperature_layer_0`.

ManuelW77, 2017-05-29:
Hmm, in my opinion it would be better to include the extruder specific code anytime.

As examples:
- I have different start scripts for left and right extruder with different coordinates where the extruder gets heated up. 
- I extrude a bit before the print starts and i have to set which extruder should do this.
- Different Nozzles for same material needs different temperatures.

maukcc, 2017-06-01:
We are also discussing this here : #1338 and #1225

ianpaschal, 2018-02-21:
No more work will be done on versions 2.x. If this is still occurring in versions 3.2 or newer, we can re-open this issue.

closed 2018-02-21 by ianpaschal

Ghostkeeper, 2018-03-13:
Since 3.2 it is now possible to specify an extruder for which a setting must be obtained in the start/end g-code, such as `{material_print_temperature, 1}` to get it from the 2nd extruder.