Ultimaker/Cura #1885: Cura 2.6 Beta - g-code printer scripts do not save

notnyt, 2017-05-25:
I edited my gcode for print completion.  It does not save.

fieldOfView, 2017-05-25:
Could you check if adding a new printer and changing the settings on that new printer works?
I think your issue is caused by a bug in the version upgrade of the old printer instance.

notnyt, 2017-05-25:
Adding a new printer to a clean install of 2.6 allows settings to be changed.  Likely a bug related to upgrading as you say.

davidedmundson, 2017-05-26:
Possibly relevant warning on opening dialog:

15:48:31.128 ?libQt5Qml.so.5?|QQmlEnginePrivate::warning|?libQt5Qml.so.5? file:///tmp/.mount_HL8KKi/usr/bin/resources/qml/Preferences/MachinesPage.qml:120: TypeError: Cannot read property 'onCompleted' of null

fieldOfView, 2017-05-26:
No, not relevant, but thanks anyway

xia0, 2017-05-26:
Old printer did not save. Added new printer, saved ok. Re-opened Cura and settings are gone again. OS X.

fieldOfView, 2017-05-27:
@xia0, could you zip up your entire configuration folder and post a link to it here?

closed 2018-03-02 by ChrisTerBeke