Ultimaker/Cura #1886: 2.6.0beta Materials issues

kr15uk, 2017-05-25:
Materials settings doesn't override some profile settings.
(works with inbuilt profiles, doesn't work with custom profiles)

I've created custom material and when selected -
- it doesn't override - hotted temp, filament diameter, retraction distance
- does override - bed temp, retraction speed, fan speed
- unable to test standby temp hence not configurable in print settings

Also regardless from the profile (custom or inbuilt) -
- in layer view mode with material colour selected it doesn't change preview to one in the Material config
- doesn't calculate cost of the print

OS - OSX 10.12.5 (clean install)

kr15uk, 2017-05-25:
Created another profile for one of my single extruder machines just to see if that's the particular profile issue.
The same issue BUT now it overrides hotted temps what wasn't happening before.

kr15uk, 2017-05-25:
Ditched dual extrusion machine from ~/Library/.../machine_instances and extruders

Colour to layer view and money calculations are back but still the same issue with overrides.

fieldOfView, 2017-05-25:
If your profiles contain a value for a setting, they will always override the value that the material sets. The idea is you don't set material diameters or temperatures in the profile.

kr15uk, 2017-05-26:
Thanks for the reply.

I've created new profile out of one of the stock profiles (Low) and haven't altered any of the settings which are in Materials.
Everything updates now but filament diameter. That thing is still broken.
Not a big game changer for the most people hence pretty much all filaments these days are spot on 1.75mm but I use quite a lot of Nylon trimmer line hence you can get them dirt cheap and diameter varies from 1.6 to 1.8...

Also problem with not updating layer view colour and cost calculations are back to normal after removing dual extruder machine and reading it back.
Probably a bug.

Guys and gals, sorry for the trouble and this ticket can now be closed.

fieldOfView, 2017-05-26:
Hmm, other than filtering, the diameter does indeed seem to be ignored. I am sorry I implied otherwise.
Well spotted, I will look into a fix.

fieldOfView, 2017-05-28:
Note that you can help test the fix, by replacing your plugins/XmlMaterialProfile/XmlMaterialProfile.py with this version:
https://github.com/fieldOfView/Cura/blob/d50ed10ffaa169106b99571890f46f8529dcb28a/plugins/XmlMaterialProfile/XmlMaterialProfile.py
(use the "raw" button to view and save the .py file)

kr15uk, 2017-05-30:
Sorry for the late update, bank holiday weekend.

Diameter update now works.
BUT it's not saving diameter changes done in materials.

Basically if I have 1.74mm material and change its diameter to 1.73mm - it's not updating settings and after Cura reboot its back to 1.74mm
Tried to update couple of materials to test, also created new one - same.
Also when you create new material it let's you set custom diameter but no changes later on.
Also reboot of Cura is required to save changes to the new material.
Unfortunately can't recollect if that was the case before.

Tried brutally setting 777 permissions on ~/Library/.../materials/* to see if that's permission issue but nope.
Also permissions for the XmlMaterialProfile.py were matched to the original prior starting Cura - mention just because we are talking permissions.

Happy to test further.

Thanks.

fieldOfView, 2017-05-30:
I have fixed a number of other issues in the same PR. Unfortunately it is now no longer possible to test this without setting up a from-source environment. For progress, see https://github.com/Ultimaker/Cura/pull/1890/

BagelOrb, 2017-07-23:
I understand that all bugs discussed here have been fixed by the ever so lovely @fieldOfView

closed 2017-07-23 by BagelOrb