Ultimaker/Cura #1887: Request: Apply infill to purge tower

tcm0116, 2017-05-26:
When printing something like a Dual Extruder Benchy on a dual extruder machine that uses the same material for a significant number of sequential layers, I noticed that the purge tower ends up containing a lot of unnecessary material. This is because the tower is being built using one material for a significant number of layers before a tool change is accomplished.

Thinking about this, it occurred to me that it would save some material (and time) if the regions within the purge tower that are all one material were treated the same as a single material object and given top/bottom layers and infill.

Ghostkeeper, 2017-08-28:
This was just rejected by our project manager for being inactive for 3 months.

jackha, 2018-08-16:
The prime tower has basically been rewritten:
- Different materials have their own "ring", inside, outside or somewhere in the middle if more than 2 extruders are used.
- It's circular.

If there are any issues with this one, we'll have a look. For now I'm closing this issue.

closed 2018-08-16 by jackha