Ultimaker/Cura #1888: Cura 2.5 on Windows 7

masklu, 2017-05-26:
when I run cura_app.py, I got this log

2017-05-26 16:22:41,428 - INFO - UM.PluginRegistry.loadPlugin [102]: Loaded plugin ConsoleLogger
2017-05-26 16:22:41,443 - INFO - CuraEngineBackend.CuraEngineBackend.__init__ [64]: Found CuraEngine at: CuraEngine.exe
2017-05-26 16:22:41,443 - INFO - UM.PluginRegistry.loadPlugin [102]: Loaded plugin CuraEngineBackend
2017-05-26 16:22:41,443 - ERROR - UM.Logger.logException [74]: Exception: Import error loading module 3MFReader
2017-05-26 16:22:41,443 - ERROR - UM.Logger.logException [78]: Traceback (most recent call last):
2017-05-26 16:22:41,443 - ERROR - UM.Logger.logException [78]:   File "C:\python35\lib\site-packages\UM\PluginRegistry.py", line 293, in _findPlugin
2017-05-26 16:22:41,443 - ERROR - UM.Logger.logException [78]:     module = imp.load_module(plugin_id, file, path, desc)
2017-05-26 16:22:41,443 - ERROR - UM.Logger.logException [78]:   File "C:\python35\lib\imp.py", line 244, in load_module
2017-05-26 16:22:41,443 - ERROR - UM.Logger.logException [78]:     return load_package(name, filename)
2017-05-26 16:22:41,443 - ERROR - UM.Logger.logException [78]:   File "C:\python35\lib\imp.py", line 216, in load_package
2017-05-26 16:22:41,443 - ERROR - UM.Logger.logException [78]:     return _load(spec)
2017-05-26 16:22:41,443 - ERROR - UM.Logger.logException [78]:   File "<frozen importlib._bootstrap>", line 693, in _load
2017-05-26 16:22:41,443 - ERROR - UM.Logger.logException [78]:   File "<frozen importlib._bootstrap>", line 673, in _load_unlocked
2017-05-26 16:22:41,443 - ERROR - UM.Logger.logException [78]:   File "<frozen importlib._bootstrap_external>", line 673, in exec_module
2017-05-26 16:22:41,443 - ERROR - UM.Logger.logException [78]:   File "<frozen importlib._bootstrap>", line 222, in _call_with_frames_removed
2017-05-26 16:22:41,443 - ERROR - UM.Logger.logException [78]:   File "C:\Cura-2.5\cura\..\plugins\3MFReader\__init__.py", line 5, in <module>
2017-05-26 16:22:41,443 - ERROR - UM.Logger.logException [78]:     from . import ThreeMFReader
2017-05-26 16:22:41,443 - ERROR - UM.Logger.logException [78]:   File "C:\Cura-2.5\cura\..\plugins\3MFReader\ThreeMFReader.py", line 23, in <module>
2017-05-26 16:22:41,443 - ERROR - UM.Logger.logException [78]:     import Savitar
2017-05-26 16:22:41,443 - ERROR - UM.Logger.logException [78]: ImportError: No module named 'Savitar'
2017-05-26 16:22:41,443 - ERROR - UM.PluginRegistry._populateMetaData [249]: Could not find plugin 3MFReader
2017-05-26 16:22:41,459 - ERROR - UM.Logger.logException [74]: Exception: Import error loading module 3MFReader
2017-05-26 16:22:41,459 - ERROR - UM.Logger.logException [78]: Traceback (most recent call last):
2017-05-26 16:22:41,459 - ERROR - UM.Logger.logException [78]:   File "C:\python35\lib\site-packages\UM\PluginRegistry.py", line 293, in _findPlugin
2017-05-26 16:22:41,459 - ERROR - UM.Logger.logException [78]:     module = imp.load_module(plugin_id, file, path, desc)
2017-05-26 16:22:41,459 - ERROR - UM.Logger.logException [78]:   File "C:\python35\lib\imp.py", line 244, in load_module
2017-05-26 16:22:41,459 - ERROR - UM.Logger.logException [78]:     return load_package(name, filename)
2017-05-26 16:22:41,459 - ERROR - UM.Logger.logException [78]:   File "C:\python35\lib\imp.py", line 216, in load_package
2017-05-26 16:22:41,459 - ERROR - UM.Logger.logException [78]:     return _load(spec)
2017-05-26 16:22:41,459 - ERROR - UM.Logger.logException [78]:   File "<frozen importlib._bootstrap>", line 693, in _load
2017-05-26 16:22:41,459 - ERROR - UM.Logger.logException [78]:   File "<frozen importlib._bootstrap>", line 673, in _load_unlocked
2017-05-26 16:22:41,459 - ERROR - UM.Logger.logException [78]:   File "<frozen importlib._bootstrap_external>", line 673, in exec_module
2017-05-26 16:22:41,459 - ERROR - UM.Logger.logException [78]:   File "<frozen importlib._bootstrap>", line 222, in _call_with_frames_removed
2017-05-26 16:22:41,459 - ERROR - UM.Logger.logException [78]:   File "C:\Cura-2.5\cura\..\plugins\3MFReader\__init__.py", line 5, in <module>
2017-05-26 16:22:41,459 - ERROR - UM.Logger.logException [78]:     from . import ThreeMFReader
2017-05-26 16:22:41,459 - ERROR - UM.Logger.logException [78]:   File "C:\Cura-2.5\cura\..\plugins\3MFReader\ThreeMFReader.py", line 23, in <module>
2017-05-26 16:22:41,459 - ERROR - UM.Logger.logException [78]:     import Savitar
2017-05-26 16:22:41,459 - ERROR - UM.Logger.logException [78]: ImportError: No module named 'Savitar'
2017-05-26 16:22:41,459 - ERROR - UM.Logger.logException [74]: Exception: Import error loading module 3MFWriter
2017-05-26 16:22:41,459 - ERROR - UM.Logger.logException [78]: Traceback (most recent call last):
2017-05-26 16:22:41,459 - ERROR - UM.Logger.logException [78]:   File "C:\python35\lib\site-packages\UM\PluginRegistry.py", line 293, in _findPlugin
2017-05-26 16:22:41,459 - ERROR - UM.Logger.logException [78]:     module = imp.load_module(plugin_id, file, path, desc)
2017-05-26 16:22:41,459 - ERROR - UM.Logger.logException [78]:   File "C:\python35\lib\imp.py", line 244, in load_module
2017-05-26 16:22:41,459 - ERROR - UM.Logger.logException [78]:     return load_package(name, filename)
2017-05-26 16:22:41,459 - ERROR - UM.Logger.logException [78]:   File "C:\python35\lib\imp.py", line 216, in load_package
2017-05-26 16:22:41,459 - ERROR - UM.Logger.logException [78]:     return _load(spec)
2017-05-26 16:22:41,459 - ERROR - UM.Logger.logException [78]:   File "<frozen importlib._bootstrap>", line 693, in _load
2017-05-26 16:22:41,459 - ERROR - UM.Logger.logException [78]:   File "<frozen importlib._bootstrap>", line 673, in _load_unlocked
2017-05-26 16:22:41,459 - ERROR - UM.Logger.logException [78]:   File "<frozen importlib._bootstrap_external>", line 673, in exec_module
2017-05-26 16:22:41,459 - ERROR - UM.Logger.logException [78]:   File "<frozen importlib._bootstrap>", line 222, in _call_with_frames_removed
2017-05-26 16:22:41,459 - ERROR - UM.Logger.logException [78]:   File "C:\Cura-2.5\cura\..\plugins\3MFWriter\__init__.py", line 6, in <module>
2017-05-26 16:22:41,459 - ERROR - UM.Logger.logException [78]:     from . import ThreeMFWriter
2017-05-26 16:22:41,459 - ERROR - UM.Logger.logException [78]:   File "C:\Cura-2.5\cura\..\plugins\3MFWriter\ThreeMFWriter.py", line 11, in <module>
2017-05-26 16:22:41,459 - ERROR - UM.Logger.logException [78]:     import Savitar
2017-05-26 16:22:41,459 - ERROR - UM.Logger.logException [78]: ImportError: No module named 'Savitar'
2017-05-26 16:22:41,459 - ERROR - UM.PluginRegistry._populateMetaData [249]: Could not find plugin 3MFWriter
2017-05-26 16:22:41,474 - ERROR - UM.Logger.logException [74]: Exception: Import error loading module 3MFWriter
2017-05-26 16:22:41,474 - ERROR - UM.Logger.logException [78]: Traceback (most recent call last):
2017-05-26 16:22:41,474 - ERROR - UM.Logger.logException [78]:   File "C:\python35\lib\site-packages\UM\PluginRegistry.py", line 293, in _findPlugin
2017-05-26 16:22:41,474 - ERROR - UM.Logger.logException [78]:     module = imp.load_module(plugin_id, file, path, desc)
2017-05-26 16:22:41,474 - ERROR - UM.Logger.logException [78]:   File "C:\python35\lib\imp.py", line 244, in load_module
2017-05-26 16:22:41,474 - ERROR - UM.Logger.logException [78]:     return load_package(name, filename)
2017-05-26 16:22:41,474 - ERROR - UM.Logger.logException [78]:   File "C:\python35\lib\imp.py", line 216, in load_package
2017-05-26 16:22:41,474 - ERROR - UM.Logger.logException [78]:     return _load(spec)
2017-05-26 16:22:41,474 - ERROR - UM.Logger.logException [78]:   File "<frozen importlib._bootstrap>", line 693, in _load
2017-05-26 16:22:41,474 - ERROR - UM.Logger.logException [78]:   File "<frozen importlib._bootstrap>", line 673, in _load_unlocked
2017-05-26 16:22:41,474 - ERROR - UM.Logger.logException [78]:   File "<frozen importlib._bootstrap_external>", line 673, in exec_module
2017-05-26 16:22:41,474 - ERROR - UM.Logger.logException [78]:   File "<frozen importlib._bootstrap>", line 222, in _call_with_frames_removed
[... 39 lines cut ...]
2017-05-26 16:22:41,693 - CRITICAL - cura.CrashHandler.show [39]:   File "C:\python35\lib\site-packages\UM\PluginRegistry.py", line 254, in _populateMetaData
2017-05-26 16:22:41,693 - CRITICAL - cura.CrashHandler.show [39]:     meta_data = plugin.getMetaData()
2017-05-26 16:22:41,693 - CRITICAL - cura.CrashHandler.show [39]: AttributeError: module 'UM' has no attribute 'getMetaData'
[... cut ...]
2017-05-26 16:22:41,708 - CRITICAL - cura.CrashHandler.show [39]:     if not self._populateMetaData(plugin_id):
2017-05-26 16:22:41,708 - CRITICAL - cura.CrashHandler.show [39]:   File "C:\python35\lib\site-packages\UM\PluginRegistry.py", line 257, in _populateMetaData
2017-05-26 16:22:41,708 - CRITICAL - cura.CrashHandler.show [39]:     raise InvalidMetaDataError(plugin_id)
2017-05-26 16:22:41,708 - CRITICAL - cura.CrashHandler.show [39]: UM.PluginError.InvalidMetaDataError: Invalid metadata for plugin UM

I've no idea to solved this

masklu, 2017-05-26:
when I compile libSavitar,I got this error:
$ mingw32-make
Scanning dependencies of target pugixml
[  5%] Building CXX object pugixml/CMakeFiles/pugixml.dir/src/pugixml.cpp.obj
[ 10%] Linking CXX static library libpugixml.a
[ 10%] Built target pugixml
Scanning dependencies of target Savitar
[ 15%] Building CXX object CMakeFiles/Savitar.dir/src/ThreeMFParser.cpp.obj
[ 21%] Building CXX object CMakeFiles/Savitar.dir/src/SceneNode.cpp.obj
[ 26%] Building CXX object CMakeFiles/Savitar.dir/src/Scene.cpp.obj
[ 31%] Building CXX object CMakeFiles/Savitar.dir/src/MeshData.cpp.obj
[ 36%] Building CXX object CMakeFiles/Savitar.dir/src/Vertex.cpp.obj
[ 42%] Building CXX object CMakeFiles/Savitar.dir/src/Face.cpp.obj
[ 47%] Linking CXX shared library libSavitar.dll
CMakeFiles\Savitar.dir/objects.a(MeshData.cpp.obj):MeshData.cpp:(.text+0x480): undefined reference to `_imp__PyBytes_FromStringAndSize'
CMakeFiles\Savitar.dir/objects.a(MeshData.cpp.obj):MeshData.cpp:(.text+0x7bc): undefined reference to `_imp__PyBytes_FromStringAndSize'
CMakeFiles\Savitar.dir/objects.a(MeshData.cpp.obj):MeshData.cpp:(.text+0x900): undefined reference to `_imp__PyBytes_FromStringAndSize'
CMakeFiles\Savitar.dir/objects.a(MeshData.cpp.obj):MeshData.cpp:(.text+0xbd2): undefined reference to `_imp__PyBytes_AsString'
CMakeFiles\Savitar.dir/objects.a(MeshData.cpp.obj):MeshData.cpp:(.text+0xbe2): undefined reference to `_imp__PyBytes_Size'
CMakeFiles\Savitar.dir/objects.a(MeshData.cpp.obj):MeshData.cpp:(.text+0xcd0): undefined reference to `_imp__PyBytes_AsString'
CMakeFiles\Savitar.dir/objects.a(MeshData.cpp.obj):MeshData.cpp:(.text+0xce0): undefined reference to `_imp__PyBytes_Size'
collect2.exe: error: ld returned 1 exit status
CMakeFiles\Savitar.dir\build.make:232: recipe for target 'libSavitar.dll' failed
mingw32-make[2]: *** [libSavitar.dll] Error 1
CMakeFiles\Makefile2:104: recipe for target 'CMakeFiles/Savitar.dir/all' failed
mingw32-make[1]: *** [CMakeFiles/Savitar.dir/all] Error 2
Makefile:128: recipe for target 'all' failed
mingw32-make: *** [all] Error 2

Ghostkeeper, 2017-05-29:
libSavitar needs to be compiled using the same compiler as your Python distribution, which in this case seems to be Microsoft Visual Studio.

Ghostkeeper, 2017-05-29:
>2017-05-26 16:22:41,693 - CRITICAL - cura.CrashHandler.show [39]: File "C:\python35\lib\site-packages\UM\PluginRegistry.py", line 254, in _populateMetaData
2017-05-26 16:22:41,693 - CRITICAL - cura.CrashHandler.show [39]: meta_data = plugin.getMetaData()
2017-05-26 16:22:41,693 - CRITICAL - cura.CrashHandler.show [39]: AttributeError: module 'UM' has no attribute 'getMetaData'

This traceback doesn't give much information :( Somehow it wants to load a plug-in called "UM". Are you sure you didn't accidentally put the Uranium source code in one of Cura's resource folders?

Also, what version are you using? In the current source code, this line of code has moved 70 lines down from where your crash report says it is.

masklu, 2017-05-31:
I use Uranium2.5 and I put the code C:\Program Files\uranium\lib\uranium\plugins into Cura's plugins folders 
if I remove those folders will log:
cura.CrashHandler.show [36]: An uncaught exception has occurred!
cura.CrashHandler.show [39]: Traceback (most recent call last):
cura.CrashHandler.show [39]:   File "C:/Cura-2.5/cura_app.py", line 70, in <module>
cura.CrashHandler.show [39]:     app = cura.CuraApplication.CuraApplication.getInstance()
cura.CrashHandler.show [39]:   File "C:\python35\lib\site-packages\UM\Application.py", line 330, in getInstance
cura.CrashHandler.show [39]:     Application._instance = cls()
cura.CrashHandler.show [39]:   File "C:\Cura-2.5\cura\CuraApplication.py", line 173, in __init__
cura.CrashHandler.show [39]:     super().__init__(name = "cura", version = CuraVersion, buildtype = CuraBuildType)
cura.CrashHandler.show [39]:   File "C:\python35\lib\site-packages\UM\Qt\QtApplication.py", line 105, in __init__
cura.CrashHandler.show [39]:     self._loadPlugins()
cura.CrashHandler.show [39]:   File "C:\Cura-2.5\cura\CuraApplication.py", line 461, in _loadPlugins
cura.CrashHandler.show [39]:     self._plugin_registry.loadPlugin("ConsoleLogger")
cura.CrashHandler.show [39]:   File "C:\python35\lib\site-packages\UM\PluginRegistry.py", line 79, in loadPlugin
cura.CrashHandler.show [39]:     raise PluginNotFoundError(plugin_id)
cura.CrashHandler.show [39]: UM.PluginError.PluginNotFoundError: Could not find plugin ConsoleLogger

fieldOfView, 2017-05-31:
Don't manually put Uranium files into Cura folders if you want to run from source. It will become a total mess.

The idea is to have the Cura and Uranium repositories in separate folders:
```
[project folder]
  Cura
    cmake
    cura
    docs
    ....
  Uranium
    cmake
    docs
    examples
    ...
  bin
    CuraEngine.exe
    libArcus.pyd
    libSavitar.pyd
```
You then set the PYTHONPATH environement variable to include the Uranium and bin folders.

closed 2017-07-23 by BagelOrb