Ultimaker/Cura #1889: Default RigidBot-Start-GCODE ommits temperature-control

aschwoerer, 2017-05-26:
Hi everyone,

The default Start-GCODE from the Rigidbot-Profile seems to be broken. There are commented Lines:

;M190 S{material_bed_temperature} ;Uncomment to add your own bed temperature line
;M109 S{material_print_temperature} ;Uncomment to add your own temperature line

that seem to prevent Cura from inserting its own values. So the resulting GCODE lacks temperature-information and therefore fails. 

Possible solutions: 
- Remove those lines from the Default-File
- Check if lines are comments before checking, if there's already an M109 or M190 defined

Greetings
Andi

nallath, 2017-05-29:
We don't do any maintenance on non UM profiles. So if anyone feels a change is necessary, make a pull request for it.

Ghostkeeper, 2017-05-31:
Well, it takes just a minute. I've removed those lines for you and the other RigidBot users: https://github.com/Ultimaker/Cura/commit/5392b8f65cf114aa873e64626bae57c1af616fd9

aschwoerer, 2017-05-31:
Thank you! As the title stated it as an rigid-bot-problem I'll close this one.

On the other hand, This problem could reoccur, if someone else uses a custom start-gcode with commented M109/M190. For me it wasnt that obvious, that Cura ommits its own code, if there is the wrong comment in the file ;)

So this could apply to users of other printers too.

closed 2017-05-31 by aschwoerer

Ghostkeeper, 2017-05-31:
Yeah, but I don't think we're going to implement better g-code parsing in that bit any time soon. We expect the designers of those definitions to be knowledgeable and test their definitions well.