Ultimaker/Cura #1892: Cura2.6 Beta - Fan speed gcode not set correct

jaybaugh, 2017-05-26:
Set "Initial Fan Speed" to 30
Set "Fan Speed" to 60
Set "Regular Fan Speed at Layer" to 2

Look at gCode, notice LAYER:0 speed set to 50%.(Why?)
;LAYER:0
M106 S127.5

Also note that fan speed is not set in layer 2 as it it supposed to be.

Ghostkeeper, 2017-05-29:
Not seeing this. I did exactly as you said, using an UM2E+ and a small-ish model. It produced this:

    ;LAYER:0
    M106 S63.8

Note that `255 * 0.25 == 63.75`, so this seems to be correct.

Then it also produces this:

    ;LAYER:1
    M106 S255

While it says `;LAYER:1`, this is because the layer numbers in the g-code count from 0 instead of 1, and Cura prefers to show layer numbers counting from 1 to the user.

jaybaugh, 2017-06-09:
Since you were not able to replicate this, I tried clearing my existing profiles and settings and tried this again and was not able to replicate it. So this is closed.

closed 2017-06-09 by jaybaugh