Ultimaker/Cura #1893: 2.5.0 Display Scaling Issues cause buttons to be inaccessible

TachyonStream, 2017-05-26:
Some menus and popups do not support display scaling in Windows 7.

The worst example is on my windows 7 PC, with display scaling to 150%, the buttons on the "Discard or Keep changes" screen (when changing a profile) are almost entirely off the bottom of the window.  It is impossible to read what the buttons are, but there is just enough left to blindly click them.  Resizing the window larger does not help, as it resizes the table and the buttons are still stuck off the screen.  An image of this is attached below.
[image]

On other screens, the buttons fit, but the text still overlaps other items (see example from post processing plugin).  This example doesn't break any functionality, but does exhibit similar symptoms.
[image]

Many of the other menus are unaffected (settings, printers, and lots of others).

Using Cura 2.5.0, on a Windows 7 PC, with display scaling at 150%.

Appesteijn, 2017-05-27:
I'm trying to fix this on 'fix_dialog_sizes' branch. After it has been tested on more machines (aka different screens) it will be merged.

ianpaschal, 2018-02-21:
No more work will be done on versions 2.x. If this is still occurring in versions 3.2 or newer, we can re-open this issue.

closed 2018-02-21 by ianpaschal

Ghostkeeper, 2018-03-13:
These should have been fixed in Cura 2.7.