Ultimaker/Cura #1896: 2.6 beta: Can't create material

ex-nerd, 2017-05-28:
On MacOS.

In the materials settings window, clicking on the `Create` button seems to do nothing.

Possibly related,  I can't duplicate materials that I have manually renamed:

* duplicate one of the existing materials
* unlink
* change some values
* exit Cura
* rename the `polymax_pla+%232.xml.fdm_material` to something like `otherbrand_pla+%232.xml.fdm_material`
* reopen cura
* select the material name corresponding to the renamed file
* press `duplicate`
* The selected material is activated, but no duplicate appears

fieldOfView, 2017-05-28:
Can you zip up your entire cura configuration folder and post it somewhere so ww can have a look?

Ghostkeeper, 2017-05-29:
And/or perhaps some of your log file?

Vandrasc, 2017-05-29:
I couldn't get so far while trying to reproduce this one, I get a crash at the _Reopen Cura_ step. occurred with an UM3, PLA material. 

{code}
Version: 2.6.0-BETA
Platform: Windows-10-10.0.14393
Qt: 5.7.1
PyQt: 5.7.1

Exception:
Traceback (most recent call last):
  File "D:/2.6/build/inst/lib/python3.5/site-packages/cura\CuraApplication.py", line 685, in event
  File "D:/2.6/build/inst/lib/python3.5/site-packages/UM\Qt\QtApplication.py", line 261, in event
  File "D:/2.6/build/inst/lib/python3.5/site-packages/UM\Event.py", line 189, in call
  File "D:/2.6/build/inst/lib/python3.5/site-packages/UM\Signal.py", line 315, in __performEmit
  File "D:/2.6/build/inst/lib/python3.5/site-packages/cura\Settings\ExtrudersModel.py", line 127, in _extrudersChanged
AttributeError: 'NoneType' object has no attribute 'containersChanged'

-------------------------
Thread 0x00001538 (most recent call first):
  File "D:/2.6/build/inst/lib/python3.5/site-packages/UM\Backend\Backend.py", line 150 in _storeStderrToLogThread
  File "D:\env\inst\Lib\threading.py", line 862 in run
  File "D:\env\inst\Lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\inst\Lib\threading.py", line 882 in _bootstrap

Thread 0x000028d4 (most recent call first):
  File "D:/2.6/build/inst/lib/python3.5/site-packages/UM\Backend\Backend.py", line 142 in _storeOutputToLogThread
  File "D:\env\inst\Lib\threading.py", line 862 in run
  File "D:\env\inst\Lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\inst\Lib\threading.py", line 882 in _bootstrap

Thread 0x00002228 (most recent call first):
  File "C:\Users\qa_tester\Desktop\CURA\Cura 2.6\plugins\USBPrinting\USBPrinterOutputDeviceManager.py", line 87 in _updateThread
  File "D:\env\inst\Lib\threading.py", line 862 in run
  File "D:\env\inst\Lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\inst\Lib\threading.py", line 882 in _bootstrap

Thread 0x00002fa4 (most recent call first):
  File "D:\env\inst\Lib\threading.py", line 293 in wait
  File "D:\env\inst\Lib\threading.py", line 424 in acquire
  File "D:/2.6/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 86 in _nextJob
  File "D:/2.6/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 116 in run
  File "D:\env\inst\Lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\inst\Lib\threading.py", line 882 in _bootstrap

Thread 0x00001a60 (most recent call first):
  File "D:\env\inst\Lib\threading.py", line 293 in wait
  File "D:\env\inst\Lib\threading.py", line 424 in acquire
  File "D:/2.6/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 86 in _nextJob
  File "D:/2.6/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 116 in run
  File "D:\env\inst\Lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\inst\Lib\threading.py", line 882 in _bootstrap

Thread 0x00002f20 (most recent call first):
  File "D:\env\inst\Lib\logging\__init__.py", line 964 in flush
  File "D:\env\inst\Lib\logging\__init__.py", line 984 in emit
  File "D:\env\inst\Lib\logging\__init__.py", line 1048 in emit
  File "D:\env\inst\Lib\logging\__init__.py", line 855 in handle
  File "D:\env\inst\Lib\logging\__init__.py", line 1487 in callHandlers
  File "D:\env\inst\Lib\logging\__init__.py", line 1425 in handle
  File "D:\env\inst\Lib\logging\__init__.py", line 1415 in _log
  File "D:\env\inst\Lib\logging\__init__.py", line 1308 in error
  File "C:\Users\qa_tester\Desktop\CURA\Cura 2.6\plugins\FileLogger\FileLogger.py", line 40 in log
  File "D:/2.6/build/inst/lib/python3.5/site-packages/UM\Logger.py", line 60 in log
  File "D:/2.6/build/inst/lib/python3.5/site-packages/UM\Logger.py", line 80 in logException
  File "D:/2.6/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 127 in run
  File "D:\env\inst\Lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\inst\Lib\threading.py", line 882 in _bootstrap

Thread 0x00001aa4 (most recent call first):
  File "D:\env\inst\Lib\threading.py", line 293 in wait
[... 43 lines cut ...]
  File "D:\env\inst\Lib\threading.py", line 882 in _bootstrap

Thread 0x00002ae8 (most recent call first):
  File "D:\env\inst\lib\site-packages\zeroconf.py", line 946 in run
  File "D:\env\inst\Lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\inst\Lib\threading.py", line 882 in _bootstrap

Thread 0x00002934 (most recent call first):
  File "D:\env\inst\Lib\threading.py", line 297 in wait
  File "D:\env\inst\lib\site-packages\zeroconf.py", line 1569 in wait
  File "D:\env\inst\lib\site-packages\zeroconf.py", line 1017 in run
  File "D:\env\inst\Lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\inst\Lib\threading.py", line 882 in _bootstrap

Thread 0x000017b0 (most recent call first):
  File "D:\env\inst\lib\site-packages\zeroconf.py", line 946 in run
  File "D:\env\inst\Lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\inst\Lib\threading.py", line 882 in _bootstrap

Current thread 0x00002d48 (most recent call first):
  File "D:/2.6/build/inst/lib/python3.5/site-packages/cura\CrashHandler.py", line 100 in show
  File "D:/2.6/build/inst/bin/cura_app.py", line 44 in exceptHook
  File "D:/2.6/build/inst/lib/python3.5/site-packages/cura\CrashHandler.py", line 116 in show
  File "D:/2.6/build/inst/bin/cura_app.py", line 44 in exceptHook
{code}

nallath, 2017-07-27:
I can't reproduce this. We made a few changes to 2.6, so I'm guessing this is fixed.

closed 2017-07-27 by nallath