Ultimaker/Cura #1899: Cura 2.6.0-Beta Crash when changing from 0.6mm Nozzle to 0.4mm Nozzle

goinowen, 2017-05-28:
Version: 2.6.0-BETA
Platform: Windows-10-10.0.15063
Qt: 5.7.1
PyQt: 5.7.1

Exception:
Traceback (most recent call last):
  File "D:/2.6/build/inst/lib/python3.5/site-packages/cura\Settings\MachineManager.py", line 788, in setActiveVariant
  File "D:/2.6/build/inst/lib/python3.5/site-packages/cura\Settings\MachineManager.py", line 769, in setActiveMaterial
  File "D:/2.6/build/inst/lib/python3.5/site-packages/cura\Settings\MachineManager.py", line 832, in setActiveQuality
  File "D:/2.6/build/inst/lib/python3.5/site-packages/cura\Settings\MachineManager.py", line 949, in _replaceQualityOrQualityChangesInStack
AttributeError: 'NoneType' object has no attribute 'getMetaDataEntry'

-------------------------
Thread 0x000012d8 (most recent call first):
  File "D:/2.6/build/inst/lib/python3.5/site-packages/UM\Backend\Backend.py", line 150 in _storeStderrToLogThread
  File "D:\env\inst\Lib\threading.py", line 862 in run
  File "D:\env\inst\Lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\inst\Lib\threading.py", line 882 in _bootstrap

Thread 0x00002cc0 (most recent call first):
  File "D:/2.6/build/inst/lib/python3.5/site-packages/UM\Backend\Backend.py", line 142 in _storeOutputToLogThread
  File "D:\env\inst\Lib\threading.py", line 862 in run
  File "D:\env\inst\Lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\inst\Lib\threading.py", line 882 in _bootstrap

Thread 0x00001c94 (most recent call first):
  File "C:\Program Files\Cura 2.6\plugins\USBPrinting\USBPrinterOutputDeviceManager.py", line 87 in _updateThread
  File "D:\env\inst\Lib\threading.py", line 862 in run
  File "D:\env\inst\Lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\inst\Lib\threading.py", line 882 in _bootstrap

Thread 0x0000249c (most recent call first):
  File "D:\env\inst\Lib\threading.py", line 293 in wait
  File "D:\env\inst\Lib\threading.py", line 424 in acquire
  File "D:/2.6/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 86 in _nextJob
  File "D:/2.6/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 116 in run
  File "D:\env\inst\Lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\inst\Lib\threading.py", line 882 in _bootstrap

Thread 0x00002910 (most recent call first):
  File "D:\env\inst\Lib\threading.py", line 293 in wait
  File "D:\env\inst\Lib\threading.py", line 424 in acquire
  File "D:/2.6/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 86 in _nextJob
  File "D:/2.6/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 116 in run
  File "D:\env\inst\Lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\inst\Lib\threading.py", line 882 in _bootstrap

Thread 0x00003adc (most recent call first):
  File "D:\env\inst\Lib\threading.py", line 293 in wait
  File "D:\env\inst\Lib\threading.py", line 424 in acquire
  File "D:/2.6/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 86 in _nextJob
  File "D:/2.6/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 116 in run
  File "D:\env\inst\Lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\inst\Lib\threading.py", line 882 in _bootstrap

Thread 0x00000f10 (most recent call first):
  File "D:\env\inst\Lib\threading.py", line 293 in wait
  File "D:\env\inst\Lib\threading.py", line 424 in acquire
  File "D:/2.6/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 86 in _nextJob
  File "D:/2.6/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 116 in run
  File "D:\env\inst\Lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\inst\Lib\threading.py", line 882 in _bootstrap

Thread 0x00002760 (most recent call first):
  File "D:\env\inst\Lib\threading.py", line 293 in wait
  File "D:\env\inst\Lib\threading.py", line 424 in acquire
  File "D:/2.6/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 86 in _nextJob
  File "D:/2.6/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 116 in run
  File "D:\env\inst\Lib\threading.py", line 914 in _bootstrap_inner
[... 90 lines cut ...]
  File "D:\env\inst\lib\site-packages\zeroconf.py", line 1569 in wait
  File "D:\env\inst\lib\site-packages\zeroconf.py", line 1017 in run
  File "D:\env\inst\Lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\inst\Lib\threading.py", line 882 in _bootstrap

Thread 0x00003100 (most recent call first):
  File "D:\env\inst\lib\site-packages\zeroconf.py", line 946 in run
  File "D:\env\inst\Lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\inst\Lib\threading.py", line 882 in _bootstrap

Current thread 0x0000344c (most recent call first):
  File "D:/2.6/build/inst/lib/python3.5/site-packages/cura\CrashHandler.py", line 100 in show
  File "D:/2.6/build/inst/bin/cura_app.py", line 44 in exceptHook
  File "D:/2.6/build/inst/lib/python3.5/site-packages/cura\CuraApplication.py", line 659 in run
  File "D:/2.6/build/inst/bin/cura_app.py", line 70 in <module>
  File "D:\env\inst\lib\site-packages\cx_Freeze\initscripts\Console.py", line 24 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 958 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 969 in _find_and_load
  File "D:\env\inst\lib\site-packages\cx_Freeze\initscripts\__startup__.py", line 12 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 958 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 969 in _find_and_load

Ghostkeeper, 2017-05-29:
I'm not able to reproduce this, using an Ultimaker 2 Extended+. Switched to 0.6mm nozzle, then back to 0.4mm. Switch went smoothly.

nallath, 2017-05-29:
Could you share some more reproduce steps?

goinowen, 2017-05-29:
Hi nallath
It happens when I have XT selected as the material. Now that I have ABS as the material I can change nozzle size OK bit it now crashes when I attempt to change the material to XT.

Ghostkeeper, 2017-05-29:
Where is this XT material you speak of?

goinowen, 2017-05-30:
It is under a Sub Menu in Materials called Colorfab and then another Sub Folder under that called XT and then the Material choice is called XT. Similar to the Ultimaker Sub Folder under materials. If I choose Ultimaker, ABS, Black ABS it doesn't crash. Also I reinstalled Cura 2.6 and allowed it to uninstall Cura 2.6 and 2.5 but that didn't help. I didn't put that material there so if it isn't in the standard install then maybe it imported it from another older version.
Thanks

Ghostkeeper, 2017-05-30:
There is no menu in Materials called Colorfab. These materials are not delivered with our distribution of Cura. Are you using an alternative distribution? Did you add that material as a custom material?

The complete list of materials we provide is here: https://github.com/Ultimaker/fdm_materials

goinowen, 2017-05-31:
Hi Ghostkeeper
I didn't enter that material in version 2.6. I may have entered it in an earlier version and it has imported it. Is that possible?
Owen

Ghostkeeper, 2017-05-31:
That's possible. It wasn't us in an older version either though.

In any case, perhaps you could provide us with that material, then. You can find it by opening Cura and then  in the application menu go to Settings -> Materials -> Manage Materials... Select the XT material and click Export. That allows you to save a file somewhere. If you put the file in a .zip archive then you can drag it in the comment box of Github here and we can download it to try to reproduce your bug :)

fieldOfView, 2017-05-31:
Wouldn't it be more effective to ask for the whole configuration folder? That way you get all the materials, the definition changes, and any possible issues caused by upgrades.

goinowen, 2017-05-31:
[XT.xml.zip](https://github.com/Ultimaker/Cura/files/1042926/XT.xml.zip)
Here you go ghostkeeper
Thanks for your help.
Owen

Ghostkeeper, 2017-05-31:
The issue reproduces when I enter the material here at home. Some insight to be gained from the log file:

    2017-05-31 23:42:42,761 - DEBUG - cura.Settings.MachineManager.setActiveMaterial [714]: Attempting to change the active material to XT_ultimaker2_plus_0.4_mm
    2017-05-31 23:42:42,771 - DEBUG - cura.Settings.MachineManager.setActiveMaterial [730]: Active material changed
    2017-05-31 23:42:42,781 - WARNING - cura.Settings.MachineManager.setActiveMaterial [764]: No quality profile found that matches the current machine and extruders.
    2017-05-31 23:42:42,791 - DEBUG - cura.Settings.MachineManager.setActiveQuality [803]: Attempting to change the active quality to ultimaker2_extended_plus_my favourites
    2017-05-31 23:42:42,801 - CRITICAL - cura.CrashHandler.show [39]: An uncaught exception has occurred!
    2017-05-31 23:42:42,826 - CRITICAL - cura.CrashHandler.show [42]: Traceback (most recent call last):
    2017-05-31 23:42:42,835 - CRITICAL - cura.CrashHandler.show [42]:   File "D:/2.6/build/inst/lib/python3.5/site-packages/cura\Settings\MachineManager.py", line 769, in setActiveMaterial
    2017-05-31 23:42:42,846 - CRITICAL - cura.CrashHandler.show [42]:   File "D:/2.6/build/inst/lib/python3.5/site-packages/cura\Settings\MachineManager.py", line 832, in setActiveQuality
    2017-05-31 23:42:42,856 - CRITICAL - cura.CrashHandler.show [42]:   File "D:/2.6/build/inst/lib/python3.5/site-packages/cura\Settings\MachineManager.py", line 949, in _replaceQualityOrQualityChangesInStack
    2017-05-31 23:42:42,865 - CRITICAL - cura.CrashHandler.show [42]: AttributeError: 'NoneType' object has no attribute 'getMetaDataEntry'

The "My Favourites" is a quality changes profile that I had active at the time.

I have not the tools to really debug it here at home. But I'll track an issue in our tracker to get this fixed.

goinowen, 2017-06-01:
OK thanks

ianpaschal, 2018-02-21:
No more work will be done on versions 2.x. If this is still occurring in versions 3.2 or newer, we can re-open this issue.

closed 2018-02-21 by ianpaschal

Ghostkeeper, 2018-03-13:
This code has been rewritten for 3.3. It should no longer occur now.

goinowen, 2018-03-13:
Cool. Thanks heaps.