Ultimaker/Cura #1900: Cura 2.5.0 Unable to scroll Dialog boxes

GregDarcy, 2017-05-29:
Cura 2.5.0
Mac OSX 10.11.1
Pointing device: Wacom Tablet INTUOS Pro L Driver version 6.3.20-8

Dialog boxes show a scroll bar when needed but they do not respond to scrolling either via the pen, or keyboard commands. Nor is it possible to resize the window. This makes it impossible to choose or even check options that are not immediately visible.

ChrisTerBeke, 2017-11-14:
Is this still an issue?

Ghostkeeper, 2017-11-24:
Yeah we probably need to make those scroll areas flickable.

ianpaschal, 2018-05-23:
As part of our new ticket handling strategy, we voted as a team on the likelihood of fixing this issue. Unfortunately, we’ve decided that we will not take time to fix it, and are marking it as `Status: Won’t Fix/Do` and closing the issue.

closed 2018-05-23 by ianpaschal

AvanOsch, 2018-06-28:
This is a bummer...
I only use my pen as pointing device, and can imagine many other people do as well.
Is this really that hard to fix? Don't know of any other software (Linux, in my case) that has scrollbar problems. I always thought the pen actually sends mouse-clicks.
Using Cura 3.4.0
Side note; using my finger as pointing device (I have a wacom pen & touch) works as expected...

fieldOfView, 2018-06-28:
I use Cura fairly exclusively with a Wacom tablet, and have not noticed an issue. Is it with all of Cura, or just certain dialogs?

One thing I do is to disable Windows Ink in the Wacom preferences, because that gives me issues with other applications. (https://sweetmonia.com/Sweet-Drawing-Blog/wacom-tablet-how-to-disable-windows-ink-keep-pressure-sensitivity-in-adobe-photoshop/)

AvanOsch, 2018-06-28:
I'm on Linux myself, but the problem is the same.
Thanks for the suggestion anyway fieldOfView.

I did some more research on this issue, and it seems to be a problem with Qt (and Wacom tablets?)...
There are lots of reports about Qt (scrollbars) not playing nice with these tablets.

I'll be looking for a workaround (might be possible by configuring differently, so the tablet is recognized as an actual mouse).
I'll report back here if I find a solution.