Ultimaker/Cura #1901: No retraction when moving over infill

notnyt, 2017-05-29:
There is currently no way to retract when moving over infill unless you set combing to off.  Doing so causes retraction between every single infill line.

smartavionics, 2017-05-29:
Uses zig-zag infill instead of lines? Increase the min travel distance to trigger a retraction?

notnyt, 2017-06-07:
And what if I want to have combing enabled?

Ghostkeeper, 2017-06-10:
Turning off combing shouldn't cause retractions when moving over infill, unless the travel move intersects with walls. Though I must admit that we rarely test slicing without combing...

Why would you want to retract when moving over infill? Does your infill need to be pretty?

notnyt, 2017-06-10:
the problem isn't that the infill needs to be pretty, it's that any filament that drags over it isn't there when it moves to print the next part.

Ghostkeeper, 2017-06-10:
Ah, so you mean that the oozed material results in underextrusion when starting the next line.

But this problem also happens with every single infill line then. What you're asking is to be able to retract when moving over infill, except when moving over infill towards other infill?

notnyt, 2017-06-10:
yes, s3d does this and it works nice

sheinz, 2017-06-17:
This option would be very useful. I also suffer from underextrusion after oozing while combing.

Ghostkeeper, 2017-09-05:
This feature was just rejected by our project manager for being more than 12 weeks old.

Toxicable, 2018-10-23:
@Ghostkeeper That dosen't sounds like a good reason for rejecting a issue.
Im also having this issue and it makes the top layer of my bottom shell poor quality due to the build up of ozzed material due to travel over infill.

smartavionics, 2018-10-23:
Cura now has an option to retract when combing if the distance is greater than some threshold value. See Travel -> Max Comb DIstance With No Retract.

Toxicable, 2018-10-23:
@smartavionics I appear to have that option set to 0, which is the default for that setting.
Yet it's definately not retracting during travel over infill

smartavionics, 2018-10-23:
I believe that the value has to be greater than 0 to enable the feature.

smartavionics, 2018-10-23:
The blurb for that setting says "When non-zero..."

Toxicable, 2018-10-23:
@smartavionics ahh... sorry, I must have mis-read it, thanks for pointing that out, will give it another shot.

notnyt, 2018-11-13:
max comb distance with no retract does not lift during infill, even on long moves.  super annoying.

[image]

smartavionics, 2018-11-15:
> max comb distance with no retract does not lift during infill, even on long moves. super annoying.

But it does retract, does it not?

notnyt, 2018-11-15:
> 
> 
> > max comb distance with no retract does not lift during infill, even on long moves. super annoying.
> 
> But it does retract, does it not?

I just looked at the g-code.  You're correct.

It does in fact retract, but no z-hop, even though it's selected.  I was assuming since no z-hop that it wasn't retracting as well.

[image]

smartavionics, 2018-11-15:
I think the lack of z-hop while combing is because UM printers z-hop very slowly and so UM have never thought it beneficial. You could open an issue and at least that would prompt a discussion about it.

notnyt, 2018-11-15:
> 
> 
> I think the lack of z-hop while combing is because UM printers z-hop very slowly and so UM have never thought it beneficial. You could open an issue and at least that would prompt a discussion about it.

Will open a new issue.. Interesting though... what exactly does that option do?  Only z-hop between parts?

smartavionics, 2018-11-15:
> Will open a new issue.. Interesting though... what exactly does that option do? Only z-hop between parts?

As far as I am aware, z-hop (if enabled) will be used for non-combed travel moves that involve retraction and also combed travel moves that move between parts.

> Also, better to open this in cura or curaengine?

Cura is fine as lots of people open issues there that are really engine issues, it makes no difference.

notnyt, 2018-11-15:
> As far as I am aware, z-hop (if enabled) will be used for non-combed travel moves that involve retraction and also combed travel moves that move between parts.
>
>
> Cura is fine as lots of people open [...]

Thanks, created new issue: https://github.com/Ultimaker/Cura/issues/4809

This is especially troublesome with this bug as well. I've been cringing every few minutes for the last couple days as the nozzle is impacting with infill on a very large print.

https://github.com/Ultimaker/CuraEngine/issues/913

[image]

mattgit, 2018-12-20:
so?

why keeping the nozzle on the print hitting it?