Ultimaker/Cura #1902: Nozzle size won't update

Naplington, 2017-05-29:
2.6.0 Beta, on Mac OS 10.12.5
Tried updating the nozzle size in the machine setup and it doesn't save. Still able to update nozzle size in Cura 2.5.0

nallath, 2017-05-29:
I just tried this with the custom machine and it worked there. Could you provide more detailed reproduction steps (and provide us with logs)

fieldOfView, 2017-05-29:
I think this only happens with upgraded printers. I have asked for entire configuration folders to check this, but sofar I got none.

@Naplington, can you zip up your configuration folder and post it somewhere so we can download it and reproduce your issue?

Also see https://github.com/Ultimaker/Cura/issues/1885

Naplington, 2017-05-29:
Excuse the very noob question - where do I find the logs?
I have only Cura 2.6.0 Beta open (although 2.5.0 is also installed but not open at the time of trying). 
I have a custom machine setup - Monoprice Mini Select
Octoprint is connected - for some reason it won’t let me disconnect Octoprint to see if the same thing happens 
I opened the printer settings and changed the nozzle diameter and pressed enter (same result if I pressed ‘close’). When I re-open the printer settings the Nozzle diameter is still 0.4

Configuration folder: https://www.dropbox.com/s/655ldkuj6w3dx73/2.6%20copy.zip?dl=0

fieldOfView, 2017-05-29:
Thanks for the configuration folder. I'm having a look at the strangeness contained within.

The OctoPrint plugin does not have a disconnect functionality. Should it? Please log an issue here:
https://github.com/fieldofview/OctoPrintPlugin/issues

Naplington, 2017-05-29:
Yeah, sorry, the configurations are a mess because I’m relatively new and learning and haven’t tidied up them up yet. Hope despite the mess it helps! 

Don’t think Octoprint has to have a disconnect, I was just going to try it as an option in case it made a difference

fieldOfView, 2017-05-29:
The strangeness is caused by Cura, so it is all on us ;-)

fieldOfView, 2017-05-30:
@Ghostkeeper could you have a look at the configuration folder supplied by @Naplington? Something gnarly happened to the fdmmaterial.xml files. They look like they were "converted" to instancecontainers containing namespaced xml strings. Wut?

Ghostkeeper, 2017-05-31:
The log is in that configuration folder that you zipped, Naplington :smiley: It's called `cura.log`.

I don't see what you mean, fieldOfView! The only available machine instance, Monoprice Mini, refers to the `custom_material` material. This `custom_material.xml.fdm_material` is properly in the `materials` folder and is a normal XML material. It's a bit strange that a custom material has all sorts of settings already pre-set, but that's fine.

I do see one mistake in the material: It should now write the specification version in the attributes of the root node:

    <fdmmaterial xmlns="http://www.ultimaker.com/material" version="1.3">

Otherwise it doesn't get loaded. Perhaps this is the source of our trouble here. It was fixed in this commit: https://github.com/Ultimaker/Cura/commit/8ede981a2c8b5a7977c83c27316193fc8a6ab5c2

fieldOfView, 2017-05-31:
@Ghostkeeper: sorry, the weirdness happens when I use the configuration folder in Naplington's archive for master (ie: rename "2.6 copy" to "master" and start Cura master). Then the .xml.fdm_material files get "upgraded" to ".inst.cfg" files, and somehow the backed up files in the "old" folder still load.

fieldOfView, 2017-05-31:
The issue at hand is with the folder ```[config folder]/2.6/machine_instances/old```. Somehow these files still get loaded by Cura 2.6, and conflict with the files they have been updated to (some of which now reside in ```[config folder]/2.6/definition_changes```.

Removing the ```old``` folder from ```machine_instances``` fixes the issue.

fieldOfView, 2017-05-31:
The problem is that VersionUpgrade25to26 upgrades definition_changes (and user, quality, quality_changes?) containers without upgrading the version (but introducing the setting_version). However Cura 2.6 only checks the version of a container, not the setting_version. So it ends up trying to load both the original container (now moved into an "old" folder) and the newly upgraded container.

Ghostkeeper, 2017-05-31:
IIRC, the upgrade should take a version that's a combination of the format-version of the container and the global setting_version. When loading, Cura should check both version numbers.

fieldOfView, 2017-05-31:
The problem is that in Cura 2.5, definition_changes InstanceContainers are in the same folder as the machine instance ContainerStacks (this is my bad, I thought that was a good idea at the time). The upgrade system rightly stores the InstanceContainers in a definition_changes folder, but stores the original files in the "old" folder. 

Cura then proceeds to load all containers, and because the old definition_changes containers are still inside the machine_instance folder they are enumerated and loaded along with machine_instances (global stacks). Since stacks don't require a setting_version (see CuraContainer.addContainer), the old definition_changes containers still get loaded.

Naplington, 2017-06-01:
I have no idea if this is relevant/helpful (because honestly don't understand most of what you have been chatting about so I may be off the mark 😄 ):

I went back to cura 2.5.0 and that would let me change the nozzle size to 0.5 and I sliced the model and printed ok. I then wondered whether that setting would be read from the same file for 2.5 and 2.6 so tried opening up 2.6 but alas nozzle was still 0.4mm. Tried restarting both and 2.5 showed 0.5 and 2.6 showed 0.4. 

Have since changed to 0.4 on all versions as I unblocked the culprit nozzle and ordered spares!

fieldOfView, 2017-06-02:
A way to fix this locally is to open the configuration folder for Cura 2.6 and remove the folder named "old" from the folder "machine_instances".

Note that once imported into Cura 2.6, the configurations of Cura 2.5 and Cura 2.6 are not synced. A change in Cura 2.5 will not reflect in Cura 2.6

fieldOfView, 2017-06-02:
It worries me that I don't see any activity on a fix, or even an internal JIRA issue nr. This happens for everybody who has opened the Machine Settings window in Cura 2.5 and upgrading to Cura 2.6, and it affects ALL settings of Machine Settings (not just nozzle size). I have tried to fix it myself, but sofar the VersionUpgrade system has proven to be black magic for me. I could also jerry-rig a fix so that Uranium simply does not try to deserialize containers whose path includes ```\/old\/\d+\/```, but I don't know if that would be accepted.

ldm314, 2017-06-03:
I am facing similar issues. Can't change the machine settings of imported printers. Trying to change the print volume, even went and modified the config file by hand. Nothing. Removed old folders. Nothing worked. I ended up removing the old profile and creating a new one.

fieldOfView, 2017-06-03:
> Removed old folders. Nothing worked.

Where did you remove them from? Note that the configuration folder for Cura 2.6 is now in AppData/Roaming instead of AppData/Local. The one in local still has your configuration from previous versions.

ldm314, 2017-06-04:
I removed the from the Local folder. I was not aware that the location had moved, I thought that it was the correct location as there was a '2.6' folder there.

Removing the 'old' folder from 'machine_instances' in the Roaming folder did solve the saving issue. I had 2 profiles imported, both showed the issue. The remaining profile I tried today showed the issue, and after removing the old folder I was able to save the changes.

awhiemstra, 2017-06-06:
> I could also jerry-rig a fix so that Uranium simply does not try to deserialize containers whose path includes \/old\/\d+\/, but I don't know if that would be accepted.

I think that is actually a good idea, that is a fairly simple fix that does not change behaviour of other parts much. Though I am not sure we really need the "old" now that we have versioned directories, but excluding it will not harm even if we remove that behaviour.

fieldOfView, 2017-06-10:
I have made a PR to Uranium to prevent loading containers from "old" folders. I hope this can still make it to the final Cura 2.6, or we'll have another iteration where upgrading does not (quite) work.

BagelOrb, 2017-07-21:
I suppose this is fixed now

closed 2017-07-21 by BagelOrb