Ultimaker/Cura #1903: Cura with text corrupted

jacques-e, 2017-05-29:
If I minimize Cura, wait some time and restore it most of the text on Cura's screen get's cripled.

Cura is installed on a VMWare machine running Windows 10 Pro updated.

[image]

Thanks in advance

Ghostkeeper, 2017-05-30:
It looks like something that could occur with only the characters that are not pixel-aligned. How's the state of your graphics drivers on that VM?

Anyways I don't suspect there is a whole lot we can do about it. Perhaps we have to align these texts better but I don't think that would help much. The rest is up to Qt.

Ghostkeeper, 2018-06-28:
IIRC we never fixed this issue but newer updates of Windows 10 don't show this problem any more. If someone coming here from a search engine is encountering this issue, try searching for Windows updates of the font rendering systems.

When a bug was occurring in an older version we won't patch the older version, but that doesn't mean that as soon as 3.4 is released all bugs reported under 3.3 are no longer a problem. They will likely still exist in the new version unless time has been spent on solving them.

ianpaschal, 2018-06-28:
> When a bug was occurring in an older version we won't patch the older version, but that doesn't mean that as soon as 3.4 is released all bugs reported under 3.3 are no longer a problem.

Of course. But there is a cut off point and as far as I'm aware our policy is that unless an old bug is reported as still occurring on a supported version (3.0 & up), we will not spend time on it.