Ultimaker/Cura #1907: Cura 2.6.9 Beta not slice even a simple model

vmzsoft, 2017-05-30:
Only show "Slicing..." and nothing.
But Cura 2.5 sliced normally.
Win7x64

smartavionics, 2017-05-30:
Although the Cura devs are clever people, their talents don't extend to mind reading. You're going to have to provide more info (images, model, profile, etc.) if there is going to be any chance of anybody understanding what the problem is here.

vmzsoft, 2017-05-30:
Model - any, even a simple little cube.
Profile - any. I tried default profile for the Ultimaker, and asked settings for my Micromake.
Why these details?
I didn't even connect the printer, I'm just interested to try new version of slicer and see the difference ...

fieldOfView, 2017-05-30:
We need logs.

Please read the readme, specifically the part about reporting issue.
https://github.com/Ultimaker/Cura/blob/master/README.md

Ghostkeeper, 2017-05-31:
I don't have your custom printer Mikromake D1, but when I load a 1x1x1cm cube in to slice for the Ultimaker 2+ it slices fine. Not able to reproduce this issue...

closed 2018-02-21 by ianpaschal