Ultimaker/Cura #1913: view profiles in preferences

maukcc, 2017-06-01:
We have set up Cura so that when a nozzle is chosen, some profiles are not being shown from the profile dropdown list. (if you have a nozzle of 0.25, there is no point in showing a profile with a layerheight of 0.6mm).

At the profiles list under preferences we would like to see all protected profiles, even if they are not supported by the current nozzle setup. This to let the user know that they are there and possible. 
Maybe even let the user know that some of them are not supported with the current machine setup. By giving them a different color or an icon.

Ghostkeeper, 2017-06-05:
This could be very confusing, since we tend to give names to profiles that are the same for all materials and variants.
You'd see 15 "Normal Quality" profiles, 15 "Fine" profiles, etc. without any indication of what the differences are between the profiles. Otherwise you'd have to make clear what material/variant each of these profiles is meant for but I don't think there is enough room in the profile manager for that.

maukcc, 2017-06-06:
It only shows the global profiles, and not the material specific ones.
So you will only see 1 fine and 1 normal and so on.

Ghostkeeper, 2017-06-10:
All quality types, you mean. We have global profiles too, but those are - consequently - not machine specific, and if there are machine specific profiles they should not be used.

Quality types have different names though, which are not intended to be read by the user, like `normal` instead of `Normal`, and `superdraft` instead of `Extra Coarse`. You could find a human-readable name by looking up any profile with that quality type and displaying that name, but that is quite a cludgy solution, especially if we ever decide to give different names to some profiles with the same quality type.

jackha, 2018-08-16:
In the current version in the recommended mode you'll see the Layer Height bar. The black part is the profiles that you can choose with the current extruder and print core combination, while the other profiles are greyed out.

[image]

closed 2018-08-16 by jackha