Ultimaker/Cura #1914: Help -> 'Show configuration folder' does nothing

mikosik, 2017-06-01:
I have cura 2.5.0 on linux manjaro.
Clicking Help -> 'Show configuration folder' does nothing. Nothing is opened, shown, displayed.
Only following line is printed to standard out:
```
thunar: symbol lookup error: /usr/lib/libpangoft2-1.0.so.0: undefined symbol: FcWeightToOpenType
```

'thunar' is default file browser on linux manjaro. It works without problems when I start it manually.

Ghostkeeper, 2017-06-05:
It's implemented here: https://github.com/Ultimaker/Cura/blob/ccec148243cffaf8b5e6b20af28653c5fc3200f9/resources/qml/Cura.qml#L772

I can't find any bugs related to this in Qt. I can't find that WeightToOpenType in Qt's source code either. Are you sure it isn't just your Linux environment that isn't properly compiled?

awhiemstra, 2017-06-06:
Are you running the AppImage? Because that runs with some changes to LD_LIBRARY_PATH which might cause this.

mikosik, 2017-06-06:
@awhiemstra Yes, I start cura via Cura-2.5.0.AppImage binary.

jackha, 2018-08-16:
Can't reproduce and no activity for a long time. Closing it

closed 2018-08-16 by jackha

Ghostkeeper, 2018-10-16:
It could've been fixed by an update of Qt. There is little else we can do about it ourselves.