Ultimaker/Cura #1917: Gradual infill improvements

hoegge, 2017-06-02:
The new gradual infill feature is really great. It would, though, make a lot more sense if you could also choose "distance to surface" as the parameter for when to half the infill, so you would get a uniform strength of your print. Not sure if that is easy to do or not

BagelOrb, 2017-06-05:
You mean to the sides and to the bottom, rather than only the top?

Wouldn't be the easiest to implement... But doable. But if you want that you should look at cubic subdivision infill pattern.

hoegge, 2017-06-07:
Thought it worked both to bottom and top. Yes - to all sides (in 3D). Similiar to Makerbot's I think

BagelOrb, 2017-06-07:
Please take a look at cubic subdivision infill.

ianpaschal, 2018-05-23:
As part of our new ticket handling strategy, we voted as a team on the likelihood of fixing this issue. Unfortunately, we’ve decided that we will not take time to fix it, and are marking it as `Status: Won’t Fix/Do` and closing the issue.

closed 2018-05-23 by ianpaschal