Ultimaker/Cura #1920: multi extruder machines prime tower and bed surface placement

maukcc, 2017-06-02:
When you have a multi extruder machine then Cura sees spaces on your build volume that are not accessible, and it should.
When you only use T0 it uses all the surface that it can.
When you only use any other extruder, it also tries to subtract the no-go-space for the prime tower, which is not used. And it puts the skirt around the (not existing) prime tower.
Cura then also subtracts the T0 no-go-spaces as well. This is not correct.

So when only 1 extruder is being used (does not matter which one), do not generate a prime tower.

Ghostkeeper, 2017-06-05:
I'm not reproducing this. Keep in mind that the platform adhesion extruder doesn't automatically change when you select all your models to print with a different extruder. While you may have selected all models to print with T1, your platform adhesion extruder might still be on T0.

maukcc, 2017-06-06:
AAh, you are correct. That was it.

Can we make this platform adhesion extruder automatic?
maybe for raft you can choose, but for skirt and brim it should be made with the extruder that is going to be used.

Ghostkeeper, 2017-06-10:
That's something I can put down as a feature request :)

Ghostkeeper, 2017-09-04:
Out project manager just removed this from our planning since it was more than 3 months old.

maukcc, 2019-09-19:
For brim this is a bug, please mark it as such, so it will get some attention :)

Ghostkeeper, 2019-09-19:
Currently this is expected behaviour even with a brim or raft. We don't expect the extruders of your print to automatically change around right now. It would be an improvement, not a bug.