Ultimaker/Cura #1921: json 'str' issue in Post-Processing Plugins

ZhuDaHai, 2017-06-02:
Using the following structure in getSettingDataString, does not function as expected:
```

                "line1":
                {
                    "label": "First Line",
                    "description": "First Inserted Line",
                    "type": "str",
                    "default_value": "This",
                    "enabled": true
                },
```

- If the field is initialized (default_value) with alpha characters, then additional alpha characters can be entered (abc / Thor / bifrost / happy)
- If field is initialized as blank (""), then only numeric characters can be entered (123 / 0.13 / 2,145)
- If field alpha character contents are deleted, then only numeric characters can be entered.
- If field is initialized with numeric characters, then only numeric characters can be entered.
- Only 10 alphanumeric characters can be entered in total into the field.

ZhuDaHai, 2017-06-02:
P.S: 
Windows 10 Professional (64bit)
Cura 2.5.0 Release

nallath, 2017-06-05:
String type is not really supported. As far as I know there is a fallback that sorta tries to make it work, but thats about it.

Ghostkeeper, 2017-06-05:
It works in the Search-And-Replace script: https://github.com/nallath/PostProcessingPlugin/blob/master/scripts/SearchAndReplace.py#L25

Ghostkeeper, 2017-06-05:
Ah, but you have to use 2.6. I made some changes to 2.6 to implement Search-and-Replace to allow this sort of thing.

closed 2018-03-02 by ChrisTerBeke