Ultimaker/Cura #1922: UM.PluginError.PluginNotFoundError: Could not find plugin ConsoleLogger

brunoaandrade, 2017-06-02:
Version: master
Platform: Mac 10.12.5
Qt: 5.9
PyQt: 5.8.2

```
An uncaught exception has occurred!
    Traceback (most recent call last):
        File "/Users/BEEVC/git/Cura/cura_app.py", line 72, in <module>
          app = cura.CuraApplication.CuraApplication.getInstance()
        File "/usr/local/lib/python3.6/site-packages/UM/Application.py", line 316, in getInstance
          Application._instance = cls()
        File "/Users/BEEVC/git/Cura/cura/CuraApplication.py", line 196, in __init__
          super().__init__(name = "cura", version = CuraVersion, buildtype = CuraBuildType)
        File "/usr/local/lib/python3.6/site-packages/UM/Qt/QtApplication.py", line 110, in __init__
          self._loadPlugins()
        File "/Users/BEEVC/git/Cura/cura/CuraApplication.py", line 490, in _loadPlugins
          self._plugin_registry.loadPlugin("ConsoleLogger")
        File "/usr/local/lib/python3.6/site-packages/UM/PluginRegistry.py", line 175, in loadPlugin
          raise PluginNotFoundError(plugin_id)
      UM.PluginError.PluginNotFoundError: Could not find plugin ConsoleLogger
```

Cura plugins but here are the installed files:
```
usr/local/lib/
/usr/local/lib//cura
/usr/local/lib//cura/plugins
/usr/local/lib//cura/plugins/3MFReader*
/usr/local/lib//cura/plugins/3MFWriter*
/usr/local/lib//cura/plugins/AutoSave*
/usr/local/lib//cura/plugins/ChangeLogPlugin*
/usr/local/lib//cura/plugins/CuraProfileReader*
/usr/local/lib//cura/plugins/GCodeProfileReader*
/usr/local/lib//cura/plugins/GCodeReader*
/usr/local/lib//cura/plugins/GCodeWriter*
/usr/local/lib//cura/plugins/ImageReader*
/usr/local/lib//cura/plugins/LayerView*
/usr/local/lib//cura/plugins/LegacyProfileReader*
/usr/local/lib//cura/plugins/MachineSettingsAction*
/usr/local/lib//cura/plugins/PerObjectSettingsTool*
/usr/local/lib//cura/plugins/RemovableDriveOutputDevice*
/usr/local/lib//cura/plugins/SliceInfoPlugin*
/usr/local/lib//cura/plugins/SolidView*
/usr/local/lib//cura/plugins/UltimakerMachineActions*
/usr/local/lib//cura/plugins/UM3NetworkPrinting*
/usr/local/lib//cura/plugins/USBPrinting*
/usr/local/lib//cura/plugins/VersionUpgrade*
/usr/local/lib//cura/plugins/X3DReader*
/usr/local/lib//cura/plugins/XmlMaterialProfile*
/usr/local/lib//cura/plugins/XRayView*

/usr/local/lib//uranium
/usr/local/lib//uranium/plugins
/usr/local/lib//uranium/plugins/ConsoleLogger*
/usr/local/lib//uranium/plugins/FileHandlers*
/usr/local/lib//uranium/plugins/FileLogger*
/usr/local/lib//uranium/plugins/LocalFileOutputDevice*
/usr/local/lib//uranium/plugins/Tools*
/usr/local/lib//uranium/plugins/UpdateChecker*
/usr/local/lib//uranium/plugins/Views*
```

smartavionics, 2017-06-02:
Hi, to run self built cura on Linux I have to use a shell script that contains this:

#!/bin/sh

PYTHONPATH=/usr/local/lib/python3/dist-packages cura $*

Maybe you have a similar problem?

nallath, 2017-06-05:
Are you sure you set all your paths correctly?

ChrisTerBeke, 2017-11-14:
Is this still an issue?

drewoprea, 2017-12-30:
I just hit this error, presumably /usr/local/lib/uranium/plugins needs to be in a path variable. I'll comment back if I figure out which one before someone tells me.

Ghostkeeper, 2018-01-17:
These are the places it should look for plug-ins when running from source on Linux:

- <UM>/../lib/uranium
- /usr/bin/plugins
- <UM>/../Resources/uranium/plugins
- <Cura>/../Resources/cura/plugins
- ~/.local/share/cura/master/plugins
- <UM>/plugins
- <Cura>/../lib/cura
- <Cura>/plugins'

mariolpantunes, 2018-03-01:
I have the same exact error with Cura 3.2.1 and running on Linux (Slackware 14.2).
I am able to fix it by moving the cura and uranium plugins to /usr/lib instead of /usr/lib64.

drewoprea, 2018-03-23:
I built cura from source because I couldn't get the .AppImage cura to properly load my printer config.
I've been busy with other work for the last two months, haven't reevaluated my problem since the 3.2.1 release.

closed 2023-05-10 by nallath