Ultimaker/Cura #1924: Settings and preferences need a confirm change sanity check

glowingwire, 2017-06-02:
For (all) software (especially of a type) that controls machines that can burn the house down, there should be an "ok/cancel" confirmation before changing settings.

People need to browse settings.
Many people have computers misconfigured to "click" when they are just moving a finger.
They don't know what the setting was... and they probably didn't want to change anything...
Fire!

Seriously, This is crazy to have preferences change with a misguided flick of a cat tail.

nallath, 2017-06-05:
We don't have preferences that can burn the machine down. 

If we would put an okay / cancel button on every setting change, i'm pretty sure the other users would lynch us, as it would make the software unusable.

Could you elaborate a bit more on what you mean?

Ghostkeeper, 2017-06-05:
We do give warnings for settings which we think are wrong, such as a print temperature of 2000 degrees.

BagelOrb, 2017-07-23:
Perhpas we do need the extra sanity check popup when you have settings in a warning state, then.

Ghostkeeper, 2017-10-30:
Our project manager just removed this from our planning because it's older than 12 weeks.

nickthetait, 2017-11-08:
This request seems pretty reasonable to me. It would only be a once-per-print job type confirmation and occurs when a users is starting a print or exporting the gcode. It would ask something like "are you sure these values are correct?" and display a list of the warning setting(s) next to their current state(s). Clicking "no" would abort the operation. Clicking "yes" would allow users to proceed like normal.

nallath, 2017-11-15:
It would add a bit of strain on the system; We'd have to check if any setting is in a warning state. Checking for the errors is already on the expensive side.

nickthetait, 2017-11-15:
I mean, only checking for error conditions.

BagelOrb, 2017-11-15:
Cura doesn't slice if there are errors. If a setting is red Cura won't generate gcode.

BagelOrb, 2017-11-15:
[image]

nickthetait, 2017-11-16:
*facepalms silently* Okay, well seems like this is a non-issue to me. Were you asking for some other kind of behavior @glowingwire ?

fieldOfView, 2017-11-16:
What we could do is add an optional “parental” lock to make inadvertent changes less likely for underprivileged users.

It is also possible to write a plugin that erases the “user” profile on exit. That reverts back any changes not saved to a profile. I would not use that plugin personally.

BagelOrb, 2017-11-17:
Currently there's no way to do a sanity check. There's no overview of what settings you change and clicking print immediately sends the gcode in case of usb or octoprint.

If you accidentally change a hidden setting when you typed something in the search bar it's even worse. You don't even see the setting.

The sanity check should list all user changes. Whether the sanity check is user initiated or always happens is up to UX.

What I currently do sometimes is change material, because then I get something close to that list of user changes, but I'm not sure it's the same thing.

fieldOfView, 2017-11-17:
You mean an overview like what you get if you press the little star icon next to the profile name when there are user settings?

BagelOrb, 2017-11-17:
Hmm I'll have to try that.. forgot that clicking the star was a special
action or whatever.

fieldOfView, 2017-11-17:
Note that it could still be useful to have that overview as an optional popup before actually printing, but I was just wondering if the information in that dialog would suffice for your needs.

glowingwire, 2017-11-23:
Really, it's very simple.
The ok/apply/cancel dialog box on windows 3.1 handles this perfectly.
KDE also does it...

gnome and macOS are built way way too complex...  I think windows10 is
starting to get way too complex...  They simplify in the wrong area, making
the problem 10,000x more complicated to solve... I think on my mac I have
to do a little wiggle and a twist and take a bow, then I can break the
system just as easily looking thru settings as before..

On (older)windows, after exploring the application, I could just hit
cancel! and I was safe!
If I saw something obvious that I wanted to change... I hit apply, and keep
poking..
If I am done, I hit OK

So freaking simple...   it directly addresses all the issues, it doesn't
require a popup just because someone has a computer with a misconfigured
trackpad (default configured is misconfigred... you look at a PC wrong and
applications are opened, files are deleted and renamed and settings are
changed.. .just because some idiot thought that accidentally touching the
trackpad should cause mayhem and provide job security for other people...)

You have to assume that your user has a pointing device that he does not
control!
(I do control mine... I had to edit the registry to make the pointing stick
work properly on the Dell , but the Lenovo didn't work at all (older
versions of their drivers did work... but I returned the laptop when they
told me that a working trackpoint was a premium support issue)

People cannot control their pointing devices...
You could record the changes of settings in transactions... but that is
probably a difficult engineering problem.

Just let people undo their changes, or make them explitly hit save... you
can warn them that settings have changed, do you want to save them... but
for the love of god make it part of the window rather than a popup.

Ghostkeeper, 2017-11-28:
We could let CuraEngineBackend's StartSliceJob also report on warning values, with a preference to disable this message?

jackha, 2018-08-16:
Closed because of inactivity

closed 2018-08-16 by jackha