Ultimaker/Cura #1925: Checksum mismatches

Hoverhead, 2017-06-02:
While 2.6.0-Beta generated code with Printerface, these messages were sent.
Faulty file attached, zipped

"Error:checksum mismatch, Last Line:4
[ERROR] Error:checksum mismatch, Last Line:4

Resend:5
echo:Unknown command:""
Error:Line Number is not Last Line Number+1, Last Line:10714
[ERROR] Error:Line Number is not Last Line Number+1, Last Line:10714

Resend:10715
o
echo:Unknown command:""
Error:Line Number is not Last Line Number+1, Last Line:24586
[ERROR] Error:Line Number is not Last Line Number+1, Last Line:24586

Resend:24587"

[CD_Support_moteur_Slic3r_concentric.gcode.zip](https://github.com/Ultimaker/Cura/files/1047679/CD_Support_moteur_Slic3r_concentric.gcode.zip)

nallath, 2017-06-05:
Who's errors are those? Those don't seem to be errors created by Cura.

Ghostkeeper, 2017-06-05:
The lines in question seem to be correct. For instance, here are lines 24586 and 24587 from your file:

    G1 X96.918 Y122.16 E493.3385
    G0 F675 X97.068 Y121.98

Looks normal. Perhaps Pronterface is doing something wrong.

ChrisTerBeke, 2017-11-14:
Is this still an issue?

Ghostkeeper, 2017-11-24:
I'm thinking this was never an issue with Cura itself.

closed 2017-11-24 by Ghostkeeper