Ultimaker/Cura #1927: Crash when opening stl file

Larks007, 2017-06-03:
Version: 2.6.0-BETA
Platform: Windows-10-10.0.14393
Qt: 5.7.1
PyQt: 5.7.1

Exception:
Traceback (most recent call last):
  File "D:/2.6/build/inst/lib/python3.5/site-packages/UM\Qt\Bindings\MainWindow.py", line 131, in mouseMoveEvent
  File "D:/2.6/build/inst/lib/python3.5/site-packages/UM\Qt\QtMouseDevice.py", line 30, in handleEvent
  File "D:/2.6/build/inst/lib/python3.5/site-packages/UM\Signal.py", line 212, in emit
  File "D:/2.6/build/inst/lib/python3.5/site-packages/UM\Signal.py", line 315, in __performEmit
  File "D:/2.6/build/inst/lib/python3.5/site-packages/UM\Controller.py", line 255, in event
  File "C:\Program Files\Cura 2.6\plugins\Tools\RotateTool\RotateTool.py", line 116, in event
    drag_start = (self.getDragStart() - handle_position).normalized()
TypeError: unsupported operand type(s) for -: 'NoneType' and 'Vector'

-------------------------
Thread 0x00003c30 (most recent call first):
  File "C:\Program Files\Cura 2.6\plugins\USBPrinting\USBPrinterOutputDeviceManager.py", line 87 in _updateThread
  File "D:\env\inst\Lib\threading.py", line 862 in run
  File "D:\env\inst\Lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\inst\Lib\threading.py", line 882 in _bootstrap

Thread 0x00003c2c (most recent call first):
  File "D:\env\inst\Lib\threading.py", line 293 in wait
  File "D:\env\inst\Lib\threading.py", line 424 in acquire
  File "D:/2.6/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 86 in _nextJob
  File "D:/2.6/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 116 in run
  File "D:\env\inst\Lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\inst\Lib\threading.py", line 882 in _bootstrap

Thread 0x00003c28 (most recent call first):
  File "D:\env\inst\Lib\threading.py", line 293 in wait
  File "D:\env\inst\Lib\threading.py", line 424 in acquire
  File "D:/2.6/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 86 in _nextJob
  File "D:/2.6/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 116 in run
  File "D:\env\inst\Lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\inst\Lib\threading.py", line 882 in _bootstrap

Thread 0x00003c24 (most recent call first):
  File "D:\env\inst\Lib\threading.py", line 293 in wait
  File "D:\env\inst\Lib\threading.py", line 424 in acquire
  File "D:/2.6/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 86 in _nextJob
  File "D:/2.6/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 116 in run
  File "D:\env\inst\Lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\inst\Lib\threading.py", line 882 in _bootstrap

Thread 0x00003c20 (most recent call first):
  File "D:\env\inst\Lib\threading.py", line 293 in wait
  File "D:\env\inst\Lib\threading.py", line 424 in acquire
  File "D:/2.6/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 86 in _nextJob
  File "D:/2.6/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 116 in run
  File "D:\env\inst\Lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\inst\Lib\threading.py", line 882 in _bootstrap

Thread 0x00003c1c (most recent call first):
  File "D:\env\inst\Lib\threading.py", line 293 in wait
  File "D:\env\inst\Lib\threading.py", line 424 in acquire
  File "D:/2.6/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 86 in _nextJob
  File "D:/2.6/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 116 in run
  File "D:\env\inst\Lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\inst\Lib\threading.py", line 882 in _bootstrap

Thread 0x00003c18 (most recent call first):
  File "D:\env\inst\Lib\threading.py", line 293 in wait
  File "D:\env\inst\Lib\threading.py", line 424 in acquire
  File "D:/2.6/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 86 in _nextJob
  File "D:/2.6/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 116 in run
  File "D:\env\inst\Lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\inst\Lib\threading.py", line 882 in _bootstrap
[... 95 lines cut ...]
  File "D:\env\inst\lib\site-packages\zeroconf.py", line 1569 in wait
  File "D:\env\inst\lib\site-packages\zeroconf.py", line 1017 in run
  File "D:\env\inst\Lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\inst\Lib\threading.py", line 882 in _bootstrap

Thread 0x00003be8 (most recent call first):
  File "D:\env\inst\lib\site-packages\zeroconf.py", line 946 in run
  File "D:\env\inst\Lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\inst\Lib\threading.py", line 882 in _bootstrap

Current thread 0x0000310c (most recent call first):
  File "D:/2.6/build/inst/lib/python3.5/site-packages/cura\CrashHandler.py", line 100 in show
  File "D:/2.6/build/inst/bin/cura_app.py", line 44 in exceptHook
  File "D:/2.6/build/inst/lib/python3.5/site-packages/cura\CuraApplication.py", line 659 in run
  File "D:/2.6/build/inst/bin/cura_app.py", line 70 in <module>
  File "D:\env\inst\lib\site-packages\cx_Freeze\initscripts\Console.py", line 24 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 958 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 969 in _find_and_load
  File "D:\env\inst\lib\site-packages\cx_Freeze\initscripts\__startup__.py", line 12 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 958 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 969 in _find_and_load

Ghostkeeper, 2017-06-05:
What STL file are you opening? Is there a way in which we can reproduce it? It obviously doesn't happen for any STL file.

Are you sure this didn't occur when dragging on something? It looks that way from the traceback.

Larks007, 2017-06-06:
I cant recall the exact file as it was a while ago and I've printed many 
things in that time.

But since it happened that one time, it has not happened since then. 
I've opened and closed countless files since then and no issue so it 
could very likely have been a problem with my computer rather than the 
software.

Thanks for getting back to me!

nallath, 2017-06-06:
Closing it as it can't be reproduced anymore.

closed 2017-06-06 by nallath