Ultimaker/Cura #1929: 2.5.0 infill settings not working with 100 % infill

Fass-ohne-Hoden, 2017-06-03:
When I set infill density to 100 %, the infill line directions setting has no effect. Instead, top/bottom line directions are used for the whole piece. Also, the whole piece is displayed in the top/bottom colour (yellow instead of orange).

Dirty fix: sett infill density to 99 % ... but, you know ... it's dirty

smartavionics, 2017-06-03:
You need to set an explicit value for "Bottom Layers" because when the infill density is 100%, bottom layers defaults to a silly large number.

Fass-ohne-Hoden, 2017-06-03:
Thanks a lot! That helped.

It is still a bug, considering I had top/bottom thickness set to a value that didn't equal 9999999 (not sure how many 9s there were) bottom layers ...

fieldOfView, 2017-06-03:
Or set infill percentage to 99

Ghostkeeper, 2017-06-05:
Or even 101%.

closed 2017-07-07 by BagelOrb