Ultimaker/Cura #1930: Cura 2.6 crashes when opening STL files

jjhartz, 2017-06-03:
I tried opening 2 separate STL files and the new Cura software crashed each time. One of the files was one that worked fine on the previous version of Cura (2.5)

The message given is:" A fatal exception has occurred that we could not recover from! Please use the information below to post a bug report at.....github."

Ghostkeeper, 2017-06-05:
Please use the information below to post a bug report at.....github.

jjhartz, 2017-06-06:
I am just a rookie at this format, so perhaps I screwed up by not posting the entire string that was spewed out by Cura when the crash occurred. Is that the issue? I cannot recreate the error condition because I uninstalled Cura 2.6 and re-installed version 2.5 which works fine. I am not really interested in being on the front lines developing the Cura 2.6 beta version. I guess I assumed that Ultimaker had done some amount of QC before release.

Ghostkeeper, 2017-06-06:
We do have three quality control layers, of which two get executed for the beta (and the third for the final release). In those QC checks we haven't seen this issue.

If we want to fix a bug here, we need to see the bug happening first. That allows us to trace where it comes from and see where it goes haywire. If it's a simple bug, the string that Cura spews out helps a lot to find it. That's why we put it there ;)

In this case, the actual STL files that you used could help as well in order to reproduce it.

jjhartz, 2017-06-06:
I believe that both the attached STL files failed with the same error message when try to run on Cura 2.6.

 

Both worked with version 2.5

 

As I mentioned, I don’t have the string since I un-installed version 2.6. Unless there is a log somewhere that I can recover that I do not know about.

Ghostkeeper, 2017-06-10:
Did you attach STL files somewhere?
Mind that replying to a Github notification e-mail with attachments will not cause the attachments to be posted online.

closed 2018-02-21 by ianpaschal