Ultimaker/Cura #1931: Layers display error on Arch Linux

sheinz, 2017-06-04:
There's a problem with showing layers in Layer Viewer on Arch Linux.
When scrolling down, layers are not shown.
There's a demo video with this issue
[https://www.youtube.com/watch?v=gb7YOO-DMB8](url)
It might be a problem of versioning as on my Ubuntu machine everything seems fine.
Or it might be related to python version, as on Arch linux python version 3 is the default whereas on Ubuntu version 2 is still a default.

I understand that it might be hard to reproduce without specific OS, but if you guide me where to look I might be able to narrow down the problem.

smartavionics, 2017-06-04:
Hi, if you enable the layer view compatibility mode in settings do you get the old layer view working OK? If so, then the problem could be related to your OpenGL implementation (drivers, libs, etc.?)

When the new layer view arrived, I had to upgrade my hardware to be able to use it.

sheinz, 2017-06-04:
@smartavionics 
In compatibility mode I get somewhat working layer view but also wrongly segmented.
It seems like the issue is related to the new dictionary implementation of Python3.6.

sheinz, 2017-06-04:
The problem was indeed in relaying on dictionary being sorted.
I made some changes to explicitly sort the layers and it fixed the issue.

Please have a look at the patch
[cura_python3.6_dict_order_fix.patch.txt](https://github.com/Ultimaker/Cura/files/1050722/cura_python3.6_dict_order_fix.patch.txt)

smartavionics, 2017-06-04:
That's great. I'm not a Python programmer so I can't really comment on what you've done but hopefully a Cura dev will pick this up. Cheers, Mark.

nallath, 2017-06-05:
@Sheinz; Could you also make a pull request? That way your code will also be included in the release.

awhiemstra, 2017-06-06:
Huh. I have seen this issue before (I'm developing on Arch) but never attributed it to Python version. Nice find!.

Note that in regards to the patch, it is probably a better idea to use collections.OrderedDict in LayerDataBuilder (replace the {} dict construction) rather than sorting when we need to render.

sheinz, 2017-06-06:
@awhiemstra Yes. My first idea was the same. To replace the regular dictionaries with OrderedDict. But in python3.6 the regular dictionaries behaves exactly like OrderedDict, they preserve order of insertion. But in our case we do not need the order of insertion. We need the layers to be sorted by key(layer number).
As far as I understand the previous versions of python keep dictionaries sorted. It was an implementation detail.

To have layers sorted I considered two approaches:
 * Explicitly declare layers as OrderedDict and sort it each time a new element is inserted.
 * Keep the dictionary as it is (for python3.6 it is just like OrderedDict) and sort it when the sorted layers are needed.

closed 2017-06-19 by nallath