Ultimaker/Cura #1932: TypeError: unable to convert a Python 'NoneType' object to a C++ 'bool' instance

greenlee, 2017-06-04:
Version: 2.6.0-BETA
Platform: Windows-10-10.0.15063
Qt: 5.7.1
PyQt: 5.7.1

Exception:
TypeError: unable to convert a Python 'NoneType' object to a C++ 'bool' instance

-------------------------
Thread 0x00003e80 (most recent call first):
  File "D:\env\inst\Lib\threading.py", line 297 in wait
  File "D:\env\inst\lib\site-packages\zeroconf.py", line 1569 in wait
  File "D:\env\inst\lib\site-packages\zeroconf.py", line 1153 in run
  File "D:\env\inst\Lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\inst\Lib\threading.py", line 882 in _bootstrap

Thread 0x00002788 (most recent call first):
  File "D:\env\inst\Lib\threading.py", line 297 in wait
  File "D:\env\inst\lib\site-packages\zeroconf.py", line 1569 in wait
  File "D:\env\inst\lib\site-packages\zeroconf.py", line 1017 in run
  File "D:\env\inst\Lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\inst\Lib\threading.py", line 882 in _bootstrap

Thread 0x00003498 (most recent call first):
  File "D:\env\inst\lib\site-packages\zeroconf.py", line 946 in run
  File "D:\env\inst\Lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\inst\Lib\threading.py", line 882 in _bootstrap

Thread 0x00000b78 (most recent call first):
  File "C:\Program Files\Cura 2.6\plugins\USBPrinting\USBPrinterOutputDeviceManager.py", line 87 in _updateThread
  File "D:\env\inst\Lib\threading.py", line 862 in run
  File "D:\env\inst\Lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\inst\Lib\threading.py", line 882 in _bootstrap

Thread 0x000017fc (most recent call first):
  File "D:\env\inst\Lib\threading.py", line 293 in wait
  File "D:\env\inst\Lib\threading.py", line 424 in acquire
  File "D:/2.6/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 86 in _nextJob
  File "D:/2.6/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 116 in run
  File "D:\env\inst\Lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\inst\Lib\threading.py", line 882 in _bootstrap

Thread 0x00003688 (most recent call first):
  File "D:\env\inst\Lib\threading.py", line 293 in wait
  File "D:\env\inst\Lib\threading.py", line 424 in acquire
  File "D:/2.6/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 86 in _nextJob
  File "D:/2.6/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 116 in run
  File "D:\env\inst\Lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\inst\Lib\threading.py", line 882 in _bootstrap

Thread 0x00003234 (most recent call first):
  File "D:\env\inst\Lib\threading.py", line 293 in wait
  File "D:\env\inst\Lib\threading.py", line 424 in acquire
  File "D:/2.6/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 86 in _nextJob
  File "D:/2.6/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 116 in run
  File "D:\env\inst\Lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\inst\Lib\threading.py", line 882 in _bootstrap

Thread 0x00002cbc (most recent call first):
  File "D:\env\inst\Lib\threading.py", line 293 in wait
  File "D:\env\inst\Lib\threading.py", line 424 in acquire
  File "D:/2.6/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 86 in _nextJob
  File "D:/2.6/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 116 in run
  File "D:\env\inst\Lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\inst\Lib\threading.py", line 882 in _bootstrap

Thread 0x00001b6c (most recent call first):
  File "D:\env\inst\Lib\threading.py", line 293 in wait
  File "D:\env\inst\Lib\threading.py", line 424 in acquire
  File "D:/2.6/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 86 in _nextJob
[... 98 lines cut ...]
  File "D:/2.6/build/inst/lib/python3.5/site-packages/cura\CrashHandler.py", line 100 in show
  File "D:/2.6/build/inst/bin/cura_app.py", line 44 in exceptHook
  File "D:/2.6/build/inst/lib/python3.5/site-packages/cura\CuraApplication.py", line 825 in updatePlatformActivity
  File "D:/2.6/build/inst/lib/python3.5/site-packages/UM\Signal.py", line 315 in __performEmit
  File "D:/2.6/build/inst/lib/python3.5/site-packages/UM\Signal.py", line 212 in emit
  File "D:/2.6/build/inst/lib/python3.5/site-packages/UM\Signal.py", line 319 in __performEmit
  File "D:/2.6/build/inst/lib/python3.5/site-packages/UM\Signal.py", line 212 in emit
  File "D:/2.6/build/inst/lib/python3.5/site-packages/UM\Signal.py", line 319 in __performEmit
  File "D:/2.6/build/inst/lib/python3.5/site-packages/UM\Signal.py", line 212 in emit
  File "D:/2.6/build/inst/lib/python3.5/site-packages/UM\Signal.py", line 319 in __performEmit
  File "D:/2.6/build/inst/lib/python3.5/site-packages/UM\Event.py", line 189 in call
  File "D:/2.6/build/inst/lib/python3.5/site-packages/UM\Qt\QtApplication.py", line 261 in event
  File "D:/2.6/build/inst/lib/python3.5/site-packages/cura\CuraApplication.py", line 685 in event
  File "D:/2.6/build/inst/lib/python3.5/site-packages/cura\CuraApplication.py", line 659 in run
  File "D:/2.6/build/inst/bin/cura_app.py", line 70 in <module>
  File "D:\env\inst\lib\site-packages\cx_Freeze\initscripts\Console.py", line 24 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 958 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 969 in _find_and_load
  File "D:\env\inst\lib\site-packages\cx_Freeze\initscripts\__startup__.py", line 12 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 958 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 969 in _find_and_load

Ghostkeeper, 2017-06-05:
Thanks for the report. What were you doing when the crash occurred? This error is fairly generic.

Unrelated, but there seem to be two threads waiting for USB devices...

greenlee, 2017-06-07:
iirc i was duplicating an object for one at a time printing.
I have a lot happening usually four different slicers open, inventor, VMs, and lots of browser tabs.

Ghostkeeper, 2017-06-10:
I don't get any errors when duplicating with one-at-a-time mode.

It gets a bit slow if I multiply a lot of objects though.

jackha, 2018-08-16:
If this is still an issue, we would have seen it.

closed 2018-08-16 by jackha