Ultimaker/Cura #1936: Wipe tower not work with 8 extruders

pablogventura, 2017-06-05:
I'm using Diamond Hotend with Cura 2.6 beta. Wipe tower not clean nozzle between T1-T7. Only work with T0, T1.

dintid, 2017-08-19:
A huge bunch of people are in need of this feature.
As it is now the wipe tower is only used for the first 2 colors used for each layer. 

I hope it's simple to simply add an/a few extra periemters to the tower for each additioanl color added?

nallath, 2017-08-21:
We don't have printers with 8 extruders, so it's impossible for us to test this. If people want to work on this, feel free to do so, but we can't.

boelle, 2017-08-21:
what is there to test? you can just look at the gcode that is generates and see if it wipes on the prime tower is using a tool above T1... no hardware is needed to test this

not really when you think of it

fieldOfView, 2017-08-21:
> you can just look at the gcode that it generates

Yes, that's called testing. 

Testing this has no benefit to Ultimaker at this point, but it would take time (and time is money). So Ultimaker will not pay its employees to test this (let alone fix it). Community members like you and I are free to test, diagnose and make fixes, but neither I nor Ultimaker has an _obligation_ to do so.

fieldOfView, 2017-08-21:
I can assure you that nowhere in the code there's something like ```if printer.brand != "Ultimaker": fuckThingsUp()". Similarly, I don't think there's a hard limit to two extruders on the wipe tower code. I do think it is highly likely that there is a bug or oversight that effectively limits the code to 2 extruders, which has not come to light because it has never been tested. But to be fair, I am not very familiar with this part of the code.

boelle, 2017-08-21:
i just had a look at the code in the cura engine code and to me its just a random bucket of code

might try and look again later and see if it makes sense. NOTE... I'm no coder/programmer :-D

boelle, 2017-08-21:
one odd thing

https://github.com/Ultimaker/CuraEngine/blob/master/src/PrimeTower.h

line 36 int extruder_count; //!< Number of extruders

why would you have a variable defined if its not in use?

i only see the variable defined on line 36 but its not used later in the file

dintid, 2017-08-22:
@nallath  @fieldOfView I can assure you both there is a limit to 2 colors pr layer usage of wipe tower in Cura. After the second it still changes tool/color, but it does not make use of the Wipe tower.. You can choose to see it as a bug or feature request, but it doesn't make sense to have this limitation, and just claiming it's because you don't have a model with 8 extruders is rather derogatory.

I'm sure Ultimaker would have liked to make a mixing hotend by now, but have had issues making it work with the existing Slizer functionality.. but inroads are clearly visibe in Cura 2.7 beta, so making hardcoded limitations now are rather counter productive. _Hence we have pointed this (bug) out to you._

Despite what Ultimaker wanted it makes sense to enable Cura Wipe Tower function to handle "infinite" number of color in order to handle future color mixing hotends. If nothing else, to support the allready existing functionality of defined Gcodes and functions in Marlin and RepRapFirmware.
In Sliz3r Prusa Edition and S3D a second Wipe just shows next to the first one when you go from 2 to 3 colors.. and so on. It's not that complicated really.

While the existing Gcodes are different for Marlin and RepRapFirmware the function is the same.

RepRapFirmware mixing function:
https://duet3d.com/wiki/G-code#M567:_Set_tool_mix_ratios

RepRap Gcode mixing function
http://reprap.org/wiki/G-code#M165:_Set_multiple_mix_weights

fieldOfView, 2017-08-22:
Having looked at the code a bit, I think I know what is going on. Nothing nefarious, I can assure you.

I can not find any check or hard limit for the first two extruders. I think if you would only print with eg extruders 4 and 7, you would get a nice wipe tower with extruders 4 and 7. There is a check if the wipe tower has already been printed for the current layer. If it has, it will not be printed again for that layer. This strategy works fine, for the scenario it has been designed for. It would need a rethink for prints with more than two extruders active on any single layer.

Having said all this, @dintid if you think this is the way to get things done or rolling, you are severely mistaken. You are biting the hand that feeds you. This will be my last message in this thread. I have said too much already.

dintid, 2017-08-22:
@fieldOfView hmm, I don't follow you on what I did to annoy you...
 I was simply stating a proper argument for, why it would be logical sound business steps for Ultimaker.

That said, I might have been a bit annoyed as you have a tendency to totally belitted any concern brought forth and act in a very derogatory term, which might have colored it somewhat.
AND as I stated elsewhere, English is not my main language, so I might easily have used some words where other synonyms shoud have been used...

dintid, 2017-08-22:
@fieldOfView but the way you describe 4 and 7 is what I tried to explain happened as welll and then I formulated argumetns for, why it wasn't a smart move in the long run.. I've never said anything about being "nefrious" or the like.. simply that I guess i wasn't thought into the code, but that it should be.

Ghostkeeper, 2017-08-31:
There is no hard-coded limit, no, but we prevent a prime tower being printed on a layer where there is already a prime tower. Since with 2 extruders you only need to swap extruders once per layer, the prime tower works with at most 2 extruders. This is the code that enforces that:

https://github.com/Ultimaker/CuraEngine/blob/79a59383b44cd5d151d69d17d5fb80a41282aacf/src/PrimeTower.cpp#L118

> line 36 int extruder_count; //!< Number of extruders
> why would you have a variable defined if its not in use?

It's actually used here: 

https://github.com/Ultimaker/CuraEngine/blob/79a59383b44cd5d151d69d17d5fb80a41282aacf/src/PrimeTower.cpp#L77

In C++, you commonly define the members of a class in a header file (`PrimeTower.h`) and the implementation in a .cpp file (`PrimeTower.cpp`).

The reason why the prime tower doesn't work for more than 2 extruders is because it needs some different techniques: either a second tower, or towers folded into each other. And what happens with the wipes and such. Just requires some thought and we haven't given it that thought.

I think eventually, given the hardware, we would implement something that fixes this. But not for the foreseeable future.

Ghostkeeper, 2017-12-04:
Our project manager just removed this issue from our planning because it's older than 12 weeks.

Ghostkeeper, 2020-04-21:
This is fixed in 3.6, where we switched the routine for the prime tower. The new routine supports >2 extruders.

closed 2020-04-21 by Ghostkeeper