Ultimaker/Cura #1938: Crash Report

REHolt, 2017-06-05:
Been using Cura Beta 2.6.0 at work all day with our Ultimaker 3+ Extended. I printed several projects. Brought My computer home this evening to set-up a new project and now I get a "Crash Report" and can't get Cura to start.

Version: 2.6.0-BETA
Platform: Windows-10-10.0.15063
Qt: 5.7.1
PyQt: 5.7.1

Exception:
Traceback (most recent call last):
  File "<frozen importlib._bootstrap>", line 969, in _find_and_load
  File "<frozen importlib._bootstrap>", line 958, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 664, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 634, in _load_backward_compatible
  File "D:\env\inst\lib\site-packages\cx_Freeze\initscripts\__startup__.py", line 12, in <module>
  File "<frozen importlib._bootstrap>", line 969, in _find_and_load
  File "<frozen importlib._bootstrap>", line 958, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 664, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 634, in _load_backward_compatible
  File "D:\env\inst\lib\site-packages\cx_Freeze\initscripts\Console.py", line 24, in <module>
  File "D:/2.6/build/inst/bin/cura_app.py", line 70, in <module>
  File "D:/2.6/build/inst/lib/python3.5/site-packages/cura\CuraApplication.py", line 630, in run
AttributeError: 'NoneType' object has no attribute 'setOrigin'

-------------------------
Thread 0x000017c0 (most recent call first):
  File "D:/2.6/build/inst/lib/python3.5/site-packages/UM\Backend\Backend.py", line 150 in _storeStderrToLogThread
  File "D:\env\inst\Lib\threading.py", line 862 in run
  File "D:\env\inst\Lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\inst\Lib\threading.py", line 882 in _bootstrap

Thread 0x0000324c (most recent call first):
  File "D:/2.6/build/inst/lib/python3.5/site-packages/UM\Backend\Backend.py", line 142 in _storeOutputToLogThread
  File "D:\env\inst\Lib\threading.py", line 862 in run
  File "D:\env\inst\Lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\inst\Lib\threading.py", line 882 in _bootstrap

Thread 0x00004d28 (most recent call first):
  File "C:\Program Files\Cura 2.6\plugins\USBPrinting\USBPrinterOutputDeviceManager.py", line 87 in _updateThread
  File "D:\env\inst\Lib\threading.py", line 862 in run
  File "D:\env\inst\Lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\inst\Lib\threading.py", line 882 in _bootstrap

Thread 0x00003cac (most recent call first):
  File "D:\env\inst\Lib\threading.py", line 293 in wait
  File "D:\env\inst\Lib\threading.py", line 424 in acquire
  File "D:/2.6/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 86 in _nextJob
  File "D:/2.6/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 116 in run
  File "D:\env\inst\Lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\inst\Lib\threading.py", line 882 in _bootstrap

Thread 0x000074b8 (most recent call first):
  File "D:\env\inst\Lib\threading.py", line 293 in wait
  File "D:\env\inst\Lib\threading.py", line 424 in acquire
  File "D:/2.6/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 86 in _nextJob
  File "D:/2.6/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 116 in run
  File "D:\env\inst\Lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\inst\Lib\threading.py", line 882 in _bootstrap

Thread 0x00001c10 (most recent call first):
  File "D:/2.6/build/inst/lib/python3.5/site-packages/UM\Job.py", line 109 in yieldThread
  File "C:\Program Files\Cura 2.6\plugins\FileHandlers\OBJReader\OBJReader.py", line 51 in read
  File "D:/2.6/build/inst/lib/python3.5/site-packages/UM\Scene\Platform.py", line 120 in run
  File "D:/2.6/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 125 in run
  File "D:\env\inst\Lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\inst\Lib\threading.py", line 882 in _bootstrap

Thread 0x00003ca8 (most recent call first):
  File "D:\env\inst\Lib\threading.py", line 293 in wait
  File "D:\env\inst\Lib\threading.py", line 424 in acquire
  File "D:/2.6/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 86 in _nextJob
[... 70 lines cut ...]
  File "D:\env\inst\lib\site-packages\zeroconf.py", line 1569 in wait
  File "D:\env\inst\lib\site-packages\zeroconf.py", line 1017 in run
  File "D:\env\inst\Lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\inst\Lib\threading.py", line 882 in _bootstrap

Thread 0x00008020 (most recent call first):
  File "D:\env\inst\lib\site-packages\zeroconf.py", line 946 in run
  File "D:\env\inst\Lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\inst\Lib\threading.py", line 882 in _bootstrap

Thread 0x00003d18 (most recent call first):
  File "D:\env\inst\Lib\threading.py", line 297 in wait
  File "D:\env\inst\lib\site-packages\zeroconf.py", line 1569 in wait
  File "D:\env\inst\lib\site-packages\zeroconf.py", line 1017 in run
  File "D:\env\inst\Lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\inst\Lib\threading.py", line 882 in _bootstrap

Thread 0x00002648 (most recent call first):
  File "D:\env\inst\lib\site-packages\zeroconf.py", line 946 in run
  File "D:\env\inst\Lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\inst\Lib\threading.py", line 882 in _bootstrap

Current thread 0x00007be0 (most recent call first):
  File "D:/2.6/build/inst/lib/python3.5/site-packages/cura\CrashHandler.py", line 100 in show
  File "D:/2.6/build/inst/bin/cura_app.py", line 44 in exceptHook

Ghostkeeper, 2017-06-06:
This seems to be the code in question: https://github.com/Ultimaker/Cura/blob/2.6/cura/CuraApplication.py#L630

So it seems that the camera couldn't be loaded. Perhaps you have a log for us? It's under `C:\Users\<you>\AppData\Roaming\cura\2.6\cura.log`. That might give some insight of why your camera tool doesn't load.

REHolt, 2017-06-08:
Sorry it took me so long but I can't seem to upload the file. it is 30mb and the systems doesn't like .log files.

REHolt, 2017-06-08:
I renamed the old .log file and re-started cura. Here is the resulting log file

[cura.txt](https://github.com/Ultimaker/Cura/files/1061059/cura.txt)

Ghostkeeper, 2017-06-10:
It says this in the log (among numerous other errors):

> 2017-06-08 08:03:09,270 - DEBUG - UM.PluginRegistry.loadPlugin [148]: Plugin CameraTool was disabled

Did you disable the camera plug-in? That shouldn't be possible.

REHolt, 2017-06-11:
No, I did not disable the camera. However, the camera has quit working.

Ghostkeeper, 2017-06-23:
There's nothing I can see directly from the log that could suggest a fix for you, and I can't reproduce this myself...

REHolt, 2017-06-23:
I installed the latest stable release (2.5?) and it works without a problem. I re-installed the 2.6 beta and the problem persists.

Ghostkeeper, 2017-07-03:
Perhaps a factory reset then? To do a factory reset, you'd remove the `C:\Users\<you>\AppData\Roaming\cura\` folder entirely (or rename it to something else such that Cura can't find it).

As a matter of fact, perhaps we can help a bit more if you put that entire folder in a .zip and upload it here, so that we can try to start Cura with your settings.

appositeit, 2017-08-17:
I had this exact problem and spent quite some time debugging it before discovering that somehow the CameraTool had been disabled in the configuration. Look at the cura.cfg per Cura Preferences and Settings Locations:

[https://github.com/Ultimaker/Cura/wiki/Cura-Preferences-and-Settings-Locations](https://github.com/Ultimaker/Cura/wiki/Cura-Preferences-and-Settings-Locations)

I had a line:

disabled_plugins = ,CameraTool

No idea how I managed that. Anyway, deleting that line removed the problem.

closed 2017-11-14 by ChrisTerBeke