Ultimaker/Cura #1939: Machine Settings Crash

Luke240, 2017-06-06:
Crashes when trying to open machine settings on a custom machine- 

Version: 2.6.0-BETA
Platform: Darwin-17.0.0-x86_64-i386-64bit
Qt: 5.7.0
PyQt: 5.7.1

Exception:
Traceback (most recent call last):
  File "/Users/ultimaker/build/2.6/build/inst/lib/python3.5/site-packages/cura/MachineAction.py", line 57, in reset
  File "/Applications/Cura.app/Contents/MacOS/plugins/plugins/MachineSettingsAction/MachineSettingsAction.py", line 73, in _reset
    self._backend.disableTimer()
AttributeError: 'NoneType' object has no attribute 'disableTimer'

-------------------------
Thread 0x000070000a1e2000 (most recent call first):
  File "/Applications/Cura.app/Contents/MacOS/plugins/plugins/USBPrinting/USBPrinterOutputDeviceManager.py", line 87 in _updateThread
  File "/Users/ultimaker/build/env/lib/python3.5/threading.py", line 862 in run
  File "/Users/ultimaker/build/env/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x0000700009cdf000 (most recent call first):
  File "/Users/ultimaker/build/env/lib/python3.5/threading.py", line 297 in wait
  File "/Users/ultimaker/build/env/lib/python3.5/site-packages/zeroconf.py", line 1569 in wait
  File "/Users/ultimaker/build/env/lib/python3.5/site-packages/zeroconf.py", line 1153 in run
  File "/Users/ultimaker/build/env/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007000097dc000 (most recent call first):
  File "/Users/ultimaker/build/env/lib/python3.5/threading.py", line 297 in wait
  File "/Users/ultimaker/build/env/lib/python3.5/site-packages/zeroconf.py", line 1569 in wait
  File "/Users/ultimaker/build/env/lib/python3.5/site-packages/zeroconf.py", line 1017 in run
  File "/Users/ultimaker/build/env/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007000092d9000 (most recent call first):
  File "/Users/ultimaker/build/env/lib/python3.5/site-packages/zeroconf.py", line 946 in run
  File "/Users/ultimaker/build/env/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x0000700008dd6000 (most recent call first):
  File "/Users/ultimaker/build/env/lib/python3.5/threading.py", line 297 in wait
  File "/Users/ultimaker/build/env/lib/python3.5/site-packages/zeroconf.py", line 1569 in wait
  File "/Users/ultimaker/build/env/lib/python3.5/site-packages/zeroconf.py", line 1017 in run
  File "/Users/ultimaker/build/env/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007000088d3000 (most recent call first):
  File "/Users/ultimaker/build/env/lib/python3.5/site-packages/zeroconf.py", line 946 in run
  File "/Users/ultimaker/build/env/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007000083d0000 (most recent call first):
  File "/Applications/Cura.app/Contents/MacOS/plugins/plugins/RemovableDriveOutputDevice/RemovableDrivePlugin.py", line 58 in _updateThread
  File "/Users/ultimaker/build/env/lib/python3.5/threading.py", line 862 in run
  File "/Users/ultimaker/build/env/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x0000700007ecd000 (most recent call first):
  File "/Users/ultimaker/build/env/lib/python3.5/threading.py", line 293 in wait
  File "/Users/ultimaker/build/env/lib/python3.5/threading.py", line 424 in acquire
  File "/Users/ultimaker/build/2.6/build/inst/lib/python3.5/site-packages/UM/JobQueue.py", line 86 in _nextJob
  File "/Users/ultimaker/build/2.6/build/inst/lib/python3.5/site-packages/UM/JobQueue.py", line 116 in run
  File "/Users/ultimaker/build/env/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007000079ca000 (most recent call first):
  File "/Users/ultimaker/build/env/lib/python3.5/threading.py", line 293 in wait
  File "/Users/ultimaker/build/env/lib/python3.5/threading.py", line 424 in acquire
  File "/Users/ultimaker/build/2.6/build/inst/lib/python3.5/site-packages/UM/JobQueue.py", line 86 in _nextJob
[... 65 lines cut ...]

Thread 0x0000700004caf000 (most recent call first):
  File "/Users/ultimaker/build/env/lib/python3.5/site-packages/zeroconf.py", line 946 in run
  File "/Users/ultimaker/build/env/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/lib/python3.5/threading.py", line 882 in _bootstrap

Current thread 0x00007ffff1846340 (most recent call first):
  File "/Users/ultimaker/build/2.6/build/inst/lib/python3.5/site-packages/cura/CrashHandler.py", line 100 in show
  File "/Users/ultimaker/build/2.6/build/inst/bin/cura_app.py", line 44 in exceptHook
  File "/Users/ultimaker/build/2.6/build/inst/lib/python3.5/site-packages/cura/CuraApplication.py", line 659 in run
  File "/Users/ultimaker/build/2.6/build/inst/bin/cura_app.py", line 70 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 958 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 969 in _find_and_load
  File "/Users/ultimaker/build/env/lib/python3.5/site-packages/cx_Freeze/initscripts/Console.py", line 21 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 958 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 969 in _find_and_load
  File "/Users/ultimaker/build/env/lib/python3.5/site-packages/cx_Freeze/initscripts/__startup__.py", line 12 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 958 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 969 in _find_and_load

nallath, 2017-06-06:
There you go :) Should be fixed now.

closed 2017-06-06 by nallath

Luke240, 2017-06-10:
Dumb question. I downloaded the dmg from your site. Is there a better way to keep up with current builds? Do I have to build from GitHub?

Ghostkeeper, 2017-06-11:
Depends on how technical you are.
* The DMG from the Ultimaker site is the only build we (the official developers) publish of Cura. The latest build there is the 2.6 beta. We typically provide a few betas and then a release every couple of months.
* There's a PPA for the Linux builds, kept by a fan, that builds daily, but that won't work on MacOS.
* Otherwise the only option is to build Cura yourself. The MacOS build is fairly challenging though.

We have the intention to provide daily builds publicly in the future.

Luke240, 2017-06-11:
Got it. I’ve been working on figuring out how to build from github on Mac. Hitting some build roadblocks but I’ll get there. Maybe.

Thanks,
Luke

commit b18b1d3c (2017-06-06): Only disable timer of backend if it's initialised

Fixes #1939