Ultimaker/Cura #1940: Some menu's not working

worksengineer, 2017-06-06:
[image]

Ghostkeeper, 2017-06-10:
Looks like it's working fine. I can click on "Set as Active Extruder" and stuff fine. Which of these menus isn't working?

regenboogviske, 2017-08-02:
I have the same issue. It's the menu's with the arrows that don't work. The first selected works, then they do not work anymore untill the menu is reopened... Difficult to describe, see screen recording in zip:

[Cura 2.6.2 - Menu's don't work.zip](https://github.com/Ultimaker/Cura/files/1193669/Cura.2.6.2.-.Menu.s.don.t.work.zip)

I'm not sure but I think this started after windows update KB4025342.

worksengineer, 2017-08-02:
[Non working menu.zip](https://github.com/Ultimaker/Cura/files/1193780/Non.working.menu.zip)

Ghostkeeper, 2017-08-18:
Did the bug resolve itself, or did you close this by accident?

worksengineer, 2017-08-18:
No its not resolved, I dont know how it was closed?

Ghostkeeper, 2017-08-29:
Perhaps you clicked on "Close and comment" instead of just "Comment" by accident.

We haven't seen this issue come up yet at work. This sort of stuff might be fixed in 2.7 though, so maybe you could try again tomorrow with the stable release of 2.7.

worksengineer, 2017-08-30:
Been using beta 2.7 and it is still the same, will download stable 2.7 and try agian.

ChrisTerBeke, 2017-11-14:
Is this still an issue?

regenboogviske, 2017-11-16:
Yes it is, I'm using the latest version of Cura but still I'm unable to select 2nd level menu items (like Modify G-Code or Check for Updates) using the mouse. As a workaround I use the arrows to select so I can do what I need to do, but not in the way I want to do it (and should be able to do it)...As this problem is linked to only one pc (Dell T1700, on my colleages pc's and laptops Cura is working fine), I asume it can be some setting/bug/driver problem/dependency/whatever in Windows?

worksengineer, 2017-11-16:
Yes this is still an issue, not changed on 3.0.4. My machine is a HP Envy i7 running Windows 10

Ghostkeeper, 2017-11-25:
> I asume it can be some setting/bug/driver problem/dependency/whatever in Windows?

Yes, we haven't seen this anywhere at our computers at work. We'll continue to keep a look out.
I haven't seen an issue about this in Qt. It's kind of hard to search for; you get all sorts of tutorials about how to make those menus in Qt.

worksengineer, 2017-11-25:
Must keep pressing the wrong button, still not resolved.

regenboogviske, 2017-11-27:
Thanks for still having a look into this.

I came upon a forum thread about Autodesk 3ds Max describing and solving a very similar issue:

https://forums.autodesk.com/t5/3ds-max-programming/3ds-max-2018-menu-problem-cant-click-on-sub-menus/td-p/7115485

The problem described here sounds exactly as what I am experiencing, be it for an other software package, but it uses qt for the menus also so this can work in a similar way? Can this be the setting we are looking for? As I'm not a programmer: how can I check this?

Stady234, 2017-12-01:
I am having a related issue on Win 10 Pro x64 / Cura 3.0.4.
First tier menu works with mouse, and you can move over the selection in second tier.  However no mouse clicks work.  I have had to use the arrow keys in first tier to scroll through second and third tier of menu items. so effectively i have to use the keyboard if i want to change anything in settings or extensions.

regenboogviske, 2017-12-01:
I think I found the solution for this, at least for my pc: for some reason the Windows register value for MenuShowDelay was set to 0 on my machine (see screenshot attached), I changed it back to the default value of 400 and now all menus in Cura are working as they should.

[image]

Ghostkeeper, 2017-12-13:
That gives a hint at what might cause this, but Cura is unable to edit that registry setting. Even if it wouldn't give a permission error in your computer, Cura shouldn't go and modify user's Windows preferences.

At least it's a workaround.

jackha, 2018-07-26:
Since there's a workaround and Cura can't do anything about it, I've closed the issue.

closed 2018-07-26 by jackha