Ultimaker/Cura #1942: Error en el inicio de cura 2.6.0

secheve, 2017-06-06:
para iniciar cura exige poner una tarjeta SD en el PC
y luego no imprime con USB si no se quita la tarjeta SD

Ghostkeeper, 2017-06-10:
¿Hablo Inglés?

Starting Cura doesn't require an SD card. What error are you getting?

You can still print via USB if there is an SD card in the computer: just click on the little arrow next to the save button.

secheve, 2017-06-11:
Starting cura, if the sdcard is not in, I get the error saying there is no sdcard.
Then I plug the card, cuts starts and then I need to remove it, otherwise I cannot print through usb

Ghostkeeper, 2017-06-19:
I can't reproduce this.

Starting Cura without SD card works fine.

Printing via USB cable works fine too. Tested using an UM2+ (and changing the g-code flavour to RepRap).

Printing via USB still works when there's an SD card inserted. But you have to click the arrow besides the print button first to select USB printing.

ChrisTerBeke, 2017-11-14:
Is this still an issue?

closed 2018-03-02 by ChrisTerBeke