Ultimaker/Cura #1946: 2.6.0 Beta  - Infill does not regenerate after changing parameters

jmagella, 2017-06-07:
The Infill in the viewport does not regenerate after changing parameters in the mac version,

nallath, 2017-06-08:
You mean in the layer view? 

Is it slicing again if you change the setting?

jmagella, 2017-06-08:
Yes in the layer view. 

It re-slices when I change settings. But the infill does not visually change. Not sure if the file itself changes. The estimated time changes so I assume the gcode has changed.

Ghostkeeper, 2017-06-10:
What setting did you change, then?

Perhaps you changed those settings on the wrong extruder? Or you override the settings in the per-mesh settings?

BagelOrb, 2017-07-21:
Maybe you disabled viewing the infill?

ChrisTerBeke, 2017-09-28:
There was a bug in the per object settings that affected the infill density if an override extruder was used. This is fixed for 3.0. Closing for now, please comment if the upcoming 3.0 release still gives you issues.

closed 2017-09-28 by ChrisTerBeke