Ultimaker/Cura #1948: Cura terminates when extruder settings get checked

maukcc, 2017-06-08:
OS Ubuntu 16.04
Cura 2.5.99 master

Cura crashes instantly when one of the following settings get checked:
Wall Extruder
Outer Wall Extruder
Inner Wall Extruder
Top/Bottom Extruder
Infill Extruder

awhiemstra, 2017-06-08:
Yes, this is a bug in PyQt that has been solved in the meantime. You need to upgrade to PyQt 5.8.2, otherwise things crash.

maukcc, 2017-06-20:
@thopiekar : is this something you can fix? or do we need to do this manually?

maukcc, 2017-06-26:
I tried installing manually. Had to 1st install pip3 then pyqt5.
Now Cura will not start any more. It will not even generate the .local/share/cura directory.
Tried to remove pyqt5 and pip3 and even purged cura* and reinstalled it, but no deal.
The appImage does work.

Ghostkeeper, 2017-06-28:
What does it say when you start a Python terminal and type:

    > import PyQt5.QtCore
    > PyQt5.QtCore.PYQT_VERSION_STR

maukcc, 2017-06-28:
I formatted my machine yesterday, so it says 5.6 now

Ghostkeeper, 2017-06-28:
That needs to become 5.8 eventually if you want to use these drop-downs.

maukcc, 2017-06-28:
Well, I tried it manually and had to format my machine. So that will not happen again :)
I will wait for @thopiekar

closed 2018-03-02 by ChrisTerBeke