Ultimaker/Cura #1949: Print time breakdown popup often refuses to appear

smartavionics, 2017-06-08:
The new popup that shows the print time breakdown is great but I am finding it often refuses to appear when I mouse over the total print time. This is on a Ubuntu Linux system that supports the new OpenGL layer view. Sometimes it appears, often not and I cannot yet determine what determines whether it will or not. The popups that appear when you mouse over the settings are all 100% reliable, it's just the time breakdown popup that's flaky.

nallath, 2017-06-08:
In that case I'm kind of afraid its a Qt issue :(

smartavionics, 2017-06-08:
Hmm, well I shall try to find out if anything in particular makes it not appear.

Ghostkeeper, 2017-06-10:
I find that it doesn't appear when the time estimate is 0. But that could be intended.

BagelOrb, 2017-07-14:
Is this still an issue?

smartavionics, 2017-07-14:
Yes, it's very unreliable.

ChrisTerBeke, 2017-11-14:
Is this still an issue?

smartavionics, 2017-11-14:
Hi,no I think it's working OK at the moment (master branch). I will close the issue.

closed 2017-11-14 by smartavionics