Ultimaker/Cura #1950: feature request : Brim on wipe tower ONLY

UltiArjan, 2017-06-08:
I often use a thicker wipe tower than needed, just to secure bed adhesion.
I would really like the option to add brim to the wipe tower, but not top the printed object.

UltiArjan, 2017-06-08:
Oh, and another option on wipe tower, i feel it makes not a lot of sense to have walls and infill for the tower. It would be faster to make it walls only.

Dim3nsioneer, 2017-06-09:
+1 This was one of the original idea of issue #1503 - in the meantime it turned out it has to be solved differently so I fully support this feature request.

Ghostkeeper, 2017-06-10:
We already had an issue about this, apparently requested by UltiArjan during the 2.4 beta (though there is no Github page in that issue in our tracker). It's in our top 50 issues, which means that it is considered important. We're getting there.

UltiArjan, 2017-06-10:
ah, good news, though is got lost somewhere, thanks for the update

Ghostkeeper, 2017-06-10:
We've tried making the prime tower consist of only walls. Or rather, we've tried concentric infill, which makes it look like walls and more importantly, prevents having to make lots of sharp corners. The problem with that is that it's hard to guarantee a certain volume of material. It's either 4 walls or 5, not 4.5. Still, in my opinion it was leagues better than the vibrator we're simulating now... But I don't know what happened to that idea. Perhaps we were too ambitious in the improvements we wanted to put in that issue.

UltiArjan, 2017-06-26:
any news on killing the vibrator?

Ghostkeeper, 2017-07-03:
The ladies seem to like it.

But owing to the shortage of ladies in our R&D department, this issue is nonetheless in our top 50. At position 20, in fact.

jackha, 2018-08-16:
We're going to make a brim for helper option.
Devs: CURA-5487

Enverex, 2019-02-01:
@jackha Have you guys made any progress on this? I'm finding tree supports are very finicky if you're not using a brim right now as it often tries to print some incredibly tiny tree bases which inevitably fail well before the print finishes, but using a brim for the whole print gives a rough edge to the actual model, so "brim on supports only" would be a perfect fix.

Ghostkeeper, 2019-02-04:
Cura 4.0 has a "Support Brim" that can be enabled separately from the normal brim.

Brim for prime tower is in the works but won't make it for 4.0. It'll be in for 4.1 most likely. It's almost finished. When that is finished we would close this ticket.

Enverex, 2019-02-04:
Thanks Ghostkeeper, so it does.

I had Cura in Expert Mode and it still wasn't showing, I had to enable "Show All Options" before it appeared - just for anyone else wondering where that option actual is.

Enverex, 2019-02-04:
Just tried, that option doesn't seem to do anything for tree supports (which are the ones that need it the most):

[image]

Enverex, 2019-02-05:
You're right, I hadn't thought to look underneath:
[image]

That said (not so much in this example, but others) there are some incredibly small trees that struggle with adhesion and I don't think even an inside brim would help (as it would only add a single line, two at most). Would it be possible to have a "minimum brim-size" for the supports so that they don't end up in that state, or do you think that may be a symptom of another issue with tree supports?

nallath, 2019-04-04:
As the feature has been implemented for 4.0, this can be closed.

closed 2019-04-04 by nallath