Ultimaker/Cura #1951: Fatal exception - beta 2.6.0 - Mac 10.12.6

jerry7171, 2017-06-08:
[Exception error.txt](https://github.com/Ultimaker/Cura/files/1061665/Exception.error.txt)

I opened an OBJ file to prepare it for printing. It suffered an exception error after I went to rotate the model.
[001.obj.zip](https://github.com/Ultimaker/Cura/files/1061673/001.obj.zip)

Ghostkeeper, 2017-06-10:
It's not giving any error when I try it (using 2.6.0 beta on Ubuntu 16.04).

Opened the same model of Maria and Jesus, then tried rotating it on every axis, just dragging the model right next to the handles instead of on top of one, rotating the intersection points of the handles, with and without snap rotation, etc. It's not crashing for me.

BagelOrb, 2017-07-14:
What OS do you have?

Ghostkeeper, 2017-07-20:
It says so right in the title: Mac 10.12.6.