Ultimaker/Cura #1954: Odd pointer vertical offset issue under hi-dpi

mqudsi, 2017-06-08:
This is a very odd one, and I'm not sure what is causing it. Today, I suddenly experienced this behavior with a Cura 2.5 beta installation that previously worked just fine, and it persisted on upgrading to 2.6 beta. Since I last used Cura, I have updated video drivers and installed several Windows Updates, it could be anything.

Basically, Cura seems to be deluding under the impression that the cursor is a cm or so higher than it actually is, making all interactions with the window click incorrectly.

Here's a video demonstrating the problem:

https://www.youtube.com/watch?v=NVxGptIR1L8&feature=youtu.be

Ghostkeeper, 2017-06-10:
Weird! Never seen this. I hesitate, but my instinct says this is a bug that's only on your system, and would be hard to find for us.

mqudsi, 2017-06-10:
I haven't either, to be honest. I don't know where I'd begin to figure this one out; I imagine in all cases it would boil down to a QT-related issue.. I should try to find another QT app and see if it reproduces.

awhiemstra, 2017-06-12:
Note that you might even get different results when testing a random Qt-based application. Most of them use QtWidgets which is quite different from QtQuick (which we use). It may not always be obvious which is which.

macman134, 2017-09-06:
I have the same issue on two separate computers, both on 2.7. The mouse is offset upward about 20-30 pixels in cura.
Tried to go back to 2.4 and the issue persists. Does anyone have a simple Qt-based program that I can run to verify its relation? I don't want to download an entire compiler just to try it once.

fieldOfView, 2017-09-06:
I would be surprised if this weren't the same issue as https://github.com/Ultimaker/Cura/issues/2068. That issue is fixed by updating to a newer Intel driver:
https://github.com/Ultimaker/Cura/issues/2068#issuecomment-324917547

macman134, 2017-09-07:
yep, that fixed it!

closed 2017-09-07 by nallath