Ultimaker/Cura #1955: Generate GCode on 2.6 beta issue

stampedeboss, 2017-06-09:
When switching between different materals the generated GCode is not picking up the proper Printing Temperature and Print Temp Initial Layer.  Regardless of the material temps set I generates 210 for both.  Bed Temp does change however.

Ghostkeeper, 2017-06-10:
Did you also take a look at the Initial Printing Temperature and the Final Printing Temperature?

Perhaps you could save a project file and upload it here so that we can try to reproduce it for you. When I'm exporting a simple cube model just now it seems to create the correct temperatures.

ethanspitz, 2017-06-11:
I also have this issue. No matter what temperature I set (I have initial and final temperature also set) it seems to want to program me down to 200.

Ghostkeeper, 2017-06-19:
Perhaps you could share a project file with us, that shows the issue. I can't reproduce it; it sets the temperature correctly to the initial layer printing temperature initially, even if I overwrite it to 206 instead of 205.

ethanspitz, 2017-06-19:
I just tried, but the save functions don't work for me either.

maukcc, 2017-06-19:
Is this on a multi extruder setup?
We had this problem as well in the start gcode. 
Because Cura does not know which extruder is going to be used 1st when inserting the start gcode it takes the default temp.

ethanspitz, 2017-06-19:
Single extruder, using custom printer though, so I'm not sure if Cura handles that as a n-extruder setup or not. (Monoprice Maker Ultimate/Wanhao Duplicator 6).

maukcc, 2017-06-19:
can you post your start_gcode?

maukcc, 2017-06-19:
It does the same for me too: 200 no matter what.
with custom printer and unchanged start gcode

maukcc, 2017-06-19:
if you add these codes to the start gcode it will work fine:
M190 S{material_bed_temperature_layer_0}
M109 S{material_print_temperature}

still: this is a bug!

maukcc, 2017-06-20:
The thing that ethanspitz is seeing is a different issue then what stampedeboss is reporting.
The ethanspitz issue is also addressed here : #1993

stampedeboss, 2017-06-20:
Actually it is the same.  It appears that it is picking up the values from the extruder profile instead of the material entries.

fieldOfView, 2017-06-22:
The issue reported by @ethanspitz and @maukcc is fixed in #2009 
@stampedeboss, are you also using a "Custom FDM Printer" with 1 extruder?

stampedeboss, 2017-06-22:
Yes

fieldOfView, 2017-06-22:
Great, then this issue is fixed with #2009

ethanspitz, 2017-06-23:
Confirmed here that #2007/#2009 fixed my issue.

closed 2017-06-23 by nallath