Ultimaker/Cura #1956: file size

nodrognosduh, 2017-06-09:
i can't open a file bigger than 1 gig.....

Ghostkeeper, 2017-06-10:
What happens when you try to open such files? Do you get an error message? Blue screen of death? Does your mouse arm get cramps? 

I just tried loading an 1.8GB file: A sphere generated in OpenSCAD using `sphere(r=40, $fn=3000);` It loaded eventually, but it took some time to get there (492s according to the log) and after loading it the frame rate dropped considerably. It then proceeded to slice in 11s. Because the slicer just collapses vertices that are too close for the resolution of the engine to handle.

nallath, 2017-06-12:
Also note that if a file is that big, it will probably contain details that get ignored by the engine in the first place.

nodrognosduh, 2017-06-12:
I just get an error message saying “can’t open file”
 
Gordon Hudson
Mech. Engineer
 

Handi Quilter, Inc.
501 North 400 West
North Salt Lake, UT 84054
Phone: 801 335-0848
Fax: 801.335.0843
www.handiquilter.com

Ghostkeeper, 2017-06-23:
Perhaps there is something in the log then. We provide more detailed error messages there.

ianpaschal, 2018-05-22:
Needs info; never got it; not reproducible, closing.

closed 2018-05-22 by ianpaschal