Ultimaker/Cura #1958: Feature request: wipe only before crossing perimeter

digiTALker, 2017-06-10:
My extruder wipes and after that it continues to print inside the same perimeter.

Ghostkeeper, 2017-06-11:
What wipe are you talking about? The wipe on the wipe tower should only happen after switching extruders.

digiTALker, 2017-06-11:
[image]
[image]
Sorry it calls Coast at End. Does`t matter with towers or not. I`m using this feature with retract to clean nozzle, an than Extra Prime Ammount.

Ghostkeeper, 2017-09-18:
Our project manager just kicked this off our planning because the ticket was older than 12 weeks.