Ultimaker/Cura #1959: Missing layer 50

tgsparky, 2017-06-10:
I have noticed with a current print that at a set layer the look of the print changes. I put the gcode into a view and found that it was missing layer 50. I ran it through Cura 2.5 and 2.6 and it did the same thing. I'm not sure if I'm doing something wrong or not. Can someone check their gcode please from cura 2.5 or 2.6

smartavionics, 2017-06-10:
Is it possible for you to make the model available? It's most likely a problem with the model.

tgsparky, 2017-06-10:
Hi Mark

I doesn’t matter what stl file I put through Cura. I have noticed something. It happens at the 15mm height mark not layer height.

smartavionics, 2017-06-10:
OK, why don't you post a screenshot of the layer view showing the missing layer. Also, you could make some gcode available that has the problem.

Ghostkeeper, 2017-06-11:
I've seen some missing layers sometimes but not always layer 50, and it would be resolved by re-slicing. We're not sure how to reproduce that...

closed 2018-03-02 by ChrisTerBeke